kamilsss655/uv-k5-firmware-custom #13: żamrażanie softu przy niektórych sygnałach

zylka5, 2023-12-26:
Witam, rewizja v0.8 miała rozwiązać problem z freezem softu, jednak tak się nie stało, jest delikatna zmiana zachowania ale problem pozostaje. W mojej okolicy na częstotliwościach powyżej 420Mhz występują ciągłe sygnały cyfrowe, jakaś energetyka czy coś. Oczywiśnie nie ma tam nic ciekawego do słuchania bo sygnał cyfrowy, ale że jest nadawany ciągle używam go do testowania anten. W wersji 0.7 było tak że jak tylko weszło się na ta częstotliwość obojętnie czy w trybie kanałowym czy vfo, radio momentalnie ulegało zamrożeniu, restart radia nie pomagał bo po włączeniu i tak pamiętało tą częstotliwość, pozostawało odkręcić antenę i zmienić częstotliwość. Na wersji 0.8 teraz nie zacina się na tej częstotliwości ale jak odstroję cie o 1kHz w dół i w górę chyba też to sytuacja się powtarza, I być może teraz potrzebny jest większy sygnał żeby doszło do konkretnego zamrożenia, Bo na małej antenie lekko przycina jak się bawię odstrajaniem, a przy zewnętrznej blokuje się konkretnie.  Nie był by to duży problem bo to tylko sygnał testowy i mogę po poprostu usunąć i znaleźć sobie inny ale zauważyłem też że podczas propagacji na falach krótkich też się to zdarza. Miałem sytuacje że podczas szukania sygnałów na paśmie 10-11m na trafiłem na jakiś sygnał jakby nośna z lekkim piskiem o wyższych częstotliwościach i też zblokowało radio na amen :D. Rano jak będzie propagacja bedę sprawdzał jak jest teraz na v0.8

kamilsss655, 2023-12-26:
Hej, dzięki za raport.
Możesz sprawdzić czy ten problem który opisałeś występuje również na sofcie egzumera czy tylko na tym?

zylka5, 2023-12-27:
Wgrywałem wcześniej soft egzumera by to sprawdzić i na zapisanym kanale problem nie wystąpił, ale nie bawiłem się w "odstarajanie" od tej częstotliwości. Może zaraz to sprawdzę i jak coś napiszę.  
U egzumer występuje podobny problem nazywany "Spectrum freezes" wielokrotnie spotało mnie to na falach krótkich,  może to ta sama przyczyna tylko u ciebie w skutek jakiś zmian przybrało na sile.

zylka5, 2023-12-27:
Nie udało mi się zamrozić przy sofcie egzumera na tym cyfrowym sygnale, ale przy okazji wykryłem inny błąd, który też nie występuje u egzumera: jak mam włączony jakiś kanał z włączonym monitorem i przejdę w tryb vfo, gdzie jest ustawiona inna częstotliwość to dalej słychać to co leci na tym kanale z przed przełączenia i trzeba kliknąć np strzałka góra/dół żeby zaktualizowało odsłuch.

zylka5, 2023-12-27:
Znalazłem sytuacje łatwą do odtworzenia na każdym radiu, na częstotliwości 26Mhz (harmoniczne może też) występuje zakłócenie z wewnętrznych generatorów i blokuje ono radio na amen, odkręcenie anteny nic nie daje bo źródło

https://github.com/kamilsss655/uv-k5-firmware-custom/assets/131492056/69c70590-c2df-4621-a2c3-11b747968147

 wewnętrzne :D

kamilsss655, 2023-12-27:
Dzięki za dokładny opis zjawiska. :)

Bardzo ciekawa sprawa z tym 26Mhz, u mnie też występuje ale poziom sygnału to -115/S2.
Dzięki Twojemu odkryciu temu mogłem zdeterminować, że ustawienia squelch (1-9) nie miały większego znaczenia na ten sygnał 26Mhz - squelch otwierał się nawet przy ustawieniu 9. Teraz zmniejszę wrażliwość squelcha i przy ustawieniu 8 u mnie 26Mhz nie otwiera już squelcha. Zmianę dodam w następnym releasie. Dzięki za testy i feedback.

Do kwestii bugu monitora/VFO jeszcze nie zajrzałem.

kamilsss655, 2023-12-28:
Nie byłem w stanie odtworzyć problemu z monitorem który opisujesz, u mnie na v.10 przejście przy monitorze z kanału na VFO powoduje zmianę odsłuchu tak jak powinno być.

closed 2023-12-28 by kamilsss655