kamilsss655/uv-k5-firmware-custom #19: VOXDel

RoadRangerX, 2023-12-30:
VOXDel = 0 seems to be about one second on my radio, VOXDel = 1 slightly longer.

kamilsss655, 2023-12-30:
Weird, for me there is no delay when `VoxDel==0`. Each `VoxDel` step is `128ms` of delay. So for 1 second of delay it would have to be set to `8`.

What `VoxSen` value you are using? Can you try with different `VoxSen` values (especially values `1` and `10`)?  The sensitivity of VOX improved, so VOX might be still open when you don't want it to be and this would have nothing to do with `VoxDel` setting.

RoadRangerX, 2023-12-31:
I tried different settings of VoxSen and VoxDel, adjusting my output to the radio's mic input until it barely transmitted the whole "burst". I'm listening to the transmission on another radio and I can see the transmit light stays on (and can hear the dead carrier) for a second after the burst stops.
I forget if the radio had a PTT delay setting that I might have set to a second and that's still active even though your firmware doesn't have a menu item for it ?

RoadRangerX, 2023-12-31:
I do see that the radio continues transmitting for maybe a half second after I release the PTT?

kamilsss655, 2023-12-31:
Not sure I can assist you further since you haven't provided the information I have asked for to reproduce the issue.

However I did the following test:

1. Disconnect any cables from the radio.
2. Set VoxDel=0
3. Set VoxSel=1
4. Tap the radio housing near the microphone a few times with your finger

Result: TX opens and closes instantly with no delay, the RED led doesn't even light up, only the "DWR" top icon changes for less than ~50ms to indicate that VOX has been activated.

**This indicates that this feature works as intended.**

RoadRangerX, 2023-12-31:
It's a K5 with a 1/4 broken screen if that matters?
Right now I have:
VOXSen 3, VOXDel 0
K5 mic gain on "+1.1dB", RxMode "Main Only", Sql "1".
I did try multiple settings.

With the cable disconnected and VOXSen=1 and VOXDel=0 tapping does nothing at all. Note that I have the mic gain at minimum.
With VOXSen=10 still doesn't respond to tapping. Then with mic=+15.1 no joy either. I can speak into it and it does key up with a bit of a delay which maybe explains why it is missing the "taps"?
I still speculate that there's some sort of PTT/VOX pre/post delay set on this radio that isn't apparent from the present menus.
I can try doing a factory reset?

Thanks, Ron

RoadRangerX, 2023-12-31:
"Reset All" didn't help :-(

RoadRangerX, 2023-12-31:
Not pertinent - but when I enable the hardware PTT on my interface (Easy Digi) it keys fine (no off delay) even with the VOX still enabled. And yes, I can see where packets from the remote station often come very quickly after I transmit one.
I'm using the Direwolf software TNC with WinLink Express.

While it's working fine with my interface, in a couple weeks I'll have the opportunity to demo the UV-K5 working with a simple $25 cable controlled by a smartphone to the entire Connecticut ARES emergency community and possibly get an article published in QST magazine - if we can get VOX working :-) . It does work occasionally now if the remote station is slow enough - as you may know packet radio stations use a random pre-delay to minimize collisions.

kamilsss655, 2024-01-01:
I'm glad you found a way to make it work.

If you're device doesn't respond to tapping following the test procedure I've provided, then your device is likely broken (you've mentioned broken screen in your unit). 

I've asked a friend to do the same procedure on a different device and we both got the same positive results. I'd suggest testing with another device, as you may know software cannot fix broken hardware.

Good luck and feel free to open an issue, if you stumble upon software issue in the future.

closed 2024-01-01 by kamilsss655

RoadRangerX, 2024-01-04:
I tried a different interface (BTECH K1 APRS cable - which is VOX only) and get similar results.
Besides "tapping" has anyone besides me tried an actual packet transmission? This radio 100% works otherwise.

I do have a Retevis RT3S running the OpenGD77 firmware that has "VOX Tail" adjustable down to 1/2 seconds, I'll see if that works better although that's still a bit on the long side. Plus it's an $80 radio.

Just to be sure - Are we talking about the same thing where your VOXDel = OpenGD77's "VOX Tail" = the time after you stop talking (audio ends) before the transmission ends?

kamilsss655, 2024-01-04:
Yup, your description of `VoxDel` is correct.
I'd get suggest testing with another unit, if the tapping doesn't work for you.

Keep me posted, and I hope it works out for you.