kamilsss655/uv-k5-firmware-custom #25: BW (Menu 2) is not stored

OE1MWW, 2024-01-02:
Changing BW to another value (for example from 25k to 12.5k) and cycle Power off/on, it will be - as in the example - 25K again.

ErikS-web, 2024-01-02:
Indeed, When power off/on it's back to 25.

Also when switching from VFO to a Memory Channel

OE1MWW, 2024-01-02:
unfortunately does this bug still exist in 0.16.2

kamilsss655, 2024-01-02:
It is not a bug.

If you want the changes made to the memory channel saved, you need to save them with menu 17 `ChSave` function.

NOTE: New filter BWs `8.33k, 6.25k, 5k` will not be saved due to CHIRP EEPROM convention that allows to store channel BW with only 1 bit, allowing to store only 2 possible BW values. In the future, once this convention changes or I modify the CHIRP driver we can store new BWs as well.

sparky93a, 2024-01-03:
ahh this relates to  my issue logged today. will see if i can close/delete #32

closed 2024-01-03 by kamilsss655