kamilsss655/uv-k5-firmware-custom #31: zawieszanie się spectrum przy niektórych sygnałach.

zylka5, 2024-01-02:
Powrócił błąd który opisywałem wcześniej ale tym razem w spectrum na ekranie dokładnego strojenia, czy jak to się tam nazywa.
I podobnie jak wcześniej na sygnałach "testowych" czyli cyfrowy i nośna na 26Mhz, zcina spectrum na ekranie szczegółowym, że przestaje reagować na klawisze i trzeba albo odkręcić antenę albo zrobić restart.

ErikS-web, 2024-01-03:
Google Translation :
The error I described earlier has returned, but this time in the spectrum on the fine tuning screen, or whatever it is called.
And just like before with the "test" signals, i.e. digital and carrier signals at 26Mhz, the spectrum on the detailed screen is cut off, it stops responding to the keys and you have to either unscrew the antenna or restart it.

You probably mean Issue #13

zylka5, 2024-01-03:
Pobawiłem się chwile podczas propagacji na zakresie 10m i podoba mi się automatyka w spectrum, sprzyja w poszukiwaniu słabych i krótkich sygnałów, Funkcja pamiętania ustawień ekranu dokładnego strojenia też jest super :+1:

kamilsss655, 2024-01-03:
> Powrócił błąd który opisywałem wcześniej ale tym razem w spectrum na ekranie dokładnego strojenia, czy jak to się tam nazywa. I podobnie jak wcześniej na sygnałach "testowych" czyli cyfrowy i nośna [...]

Hmm powiem szczerze, że nie jestem w stanie odtworzyć tego problemu. Jakie masz ustawienie squelch?

Spróbuj ustawić squelch na 9 i zobacz czy się wtedy zatnie.

zylka5, 2024-01-03:
A to poziom sql z menu ma znaczenie w spectrum, akurat miałem wysoko na 8.  To podam więcej szczegółów jak odtworzyć efekt. Na poziom sygnału tej wewnętrznej nośnej ma trochę wpływ rodzaj lub kierunek podłączonej anteny i czasami może nie wystąpić. Wiec można odłączyć całkiem antenę, wpisujemy w vfo 26,0Mhz, wchodzimy w spectrum, zjeżdżamy spectrumowym sql czyli tą kreską pod przyciskiem F aż zacznie szumieć, wciskamy klawisz M żeby wejść na dokładne strojenie i  jak nie zawiśnie od razu, to przestrajamy trochę w okolicach kilkanaście kiloherców i powinno zawisnąć. Np teraz zwiecha na 26,0065

zylka5, 2024-01-03:
Ja tam bym chciał mieć jak bardziej czułe radio względem słabych i odległych sygnałów, ostatnio słyszy się coraz więcej sygnałów, że soft od egzumer jest głuchy i przydało by się regulacja RX gain coś jak w sofcie ijv. 
Jeśli chodzi o squelcha to dla mnie to służy do "odcinania" szumów, gdy nie ma pożądanego sygnału do odsłuchu. Ułatwia to komfortowe użytkowanie krótkofalówki. Nie rozumiem co to ma wspólnego z silnymi sygnałami ani dlaczego miały by zawieszać softa i dlaczego tylko nie które. To by byle sygnał jak toś jest w pobliżu powinien powodować ten efekt.  I jeśli było grzebane tylko w nim i powróciło do poprzedniego poziomu to czemu teraz problem nie występuje ponownie w vfo (i to bez dzwieku ) tylko w spectrum i to tylko na ekranie dokładnego strojenia  (i to z "dźwiekiem" ).  W zasadzie to mi jakoś nie przeszkadza bo wiem kiedy i jakich częstotliwości unikać a jak wystąpi to szybko sobie z nim poradzę.

kamilsss655, 2024-01-04:
Odnośnie tych porównań co były ostatnio, to nie były one wykonane porządnie więc nie biorę ich na poważnie. U @egzumer ustawili większy filtr i nie wiadomo jakie były ustawienia gaina (a mogłbyć niższy bo np. ktoś używał wcześniej modulacji AM). Więc te porównania są nierzetelne. Ja bazuję na egzumerze i mam bardzo dobry odbiór nawet w paśmie HF. Inni użytkownicy też raportowali, że nie widzą różnicy między tym fw a IJV jeżeli chodzi o wrażliwość.

Odnośnie squelcha nie wiem dlaczego, ale tak jest że chroni też radio przed silnymi sygnałami. W spectrum jak włączysz dzwięk to niejako wyłączasz to zabezpieczenie i stąd być może efekt jaki opisałeś.

Myślę, że możemy zamknąć ten issue?

zylka5, 2024-01-04:
skoro temat wyczerpany można zamykać.

kamilsss655, 2024-01-04:
Ok, dawaj znać jakby wpadło coś nowego.

closed 2024-01-04 by kamilsss655