kamilsss655/uv-k5-firmware-custom #37: scan wait time with 'CARRIER' set in Menu 23

OE1MWW, 2024-01-04:
On some services, during scan, the time of 3.6 seconds is to short to hear the response
on the current channel. Scan will continue after 3.6 seconds if station ends transmission, 
one will miss the rest of the conversation. As I can compile a modified version in codespace, 
this is not a big deal for me.

Changed "const uint16_t    scan_pause_delay_in_7_10ms       =  3600 / 10;   // 3.6 seconds"
in the file 'misc.c' from 3600 to 6000 = 6 seconds

Maybe an optional manual entry for that in the menu would help others too.

zylka5, 2024-01-05:
Mi też przeszkadza tak krótki czas oczekiwania i czasami zwiększam sobie tą wartość na sztywno w kodzie.
Na tą dolegliwość mam ciekawe rozwiązanie, mianowicie kanał na którym ostatnio zeskanowano transmisje wskakiwał by jako kanał priorytetowy na okres np. 3min. W ten sposób nie ucinało my kontynuacji rozmowy i jednocześnie kontynuowało by się skanowanie w razie jak by miała nie wystąpić kontynuacja (2 kanały priorytetowe są wbudowane w soft i można je ustawiać na sztywno w programie chirp, takie dynamiczne kanały priorytetowe wydaje mi się że były by ciekawym rozwiązaniem).

kamilsss655, 2024-01-05:
Please note that each increase of wait time, lowers the ability to receive other signals. This is a tradeoff I am not willing to take at this point in time, so its best to compile your own modified version that suits your needs.

For me personally this wait time is a bit too long, so opinions and use cases might vary.

Perhaps if I get more requests in the future I might change it.

closed 2024-01-05 by kamilsss655