kamilsss655/uv-k5-firmware-custom #41: Is this needed - TX end of transmission fix

kamilsss655, 2024-01-05:
> Additionally, gFlagEndTransmission == true, which was being set when
a serial config was in progress or when TX timeout (but not always...(?))
path was switching to foreground function w/o stopping the radio, and
some RF pulses were emited after the radio was kicked out from TX mode
upon, for instance, the beginning of a serial configuration process.

Wondering if this fix should be imported: https://github.com/joaquimorg/uv-k5-firmware-custom/commit/3b96b2ad32518b8464bd2d2a9c3d853c50b0813d

cc @ ErikS-web

ErikS-web, 2024-01-05:
Any idea how to test old situation, as I never noticed it ?

kamilsss655, 2024-01-05:
Wondering if this means that radio transmits when not wanted? Can you do couple of tests like CHIRP programming and receive on the active TX frequency to make sure it doesn't transmit?

Also can you check it transmit timeout works properly?

ErikS-web, 2024-01-05:
I am still wondering how to detect a tinx TX. Does it turns Red LED on ?

I will keep my powermeter attatched for some while. But if analog meter is fast enough ...

kamilsss655, 2024-01-05:
I'd test while setting 2 radios on the same channel. And do bunch of tests to see if the radio transmits when it shouldn't.

kamilsss655, 2024-01-09:
Closing for now.

closed 2024-01-09 by kamilsss655