kamilsss655/uv-k5-firmware-custom #44: Zawieszanie sie spectrum w trybie kanałowym

zylka5, 2024-01-06:
W trybie kanałowym czyli hiper szybkim skanowaniu kanałów, sprctrum zawiesza się natychmiast po wejściu. Nic nie da się zrobić trzeba wyłączyć radio.

kamilsss655, 2024-01-07:
Obniż gain na 0 z odłączoną anteną i dopiero podłącz. 
Zwiększaj gain tylko do takiej wartości, przy której nie występuje zawieszanie.
Tak jak pisałem to radio nie jest odporne na mocne sygnały, szczególnie w spectrum gdzie nie chroni go squelch.

zylka5, 2024-01-07:
Ale gdzie gain jak? 
Dziwne dziś działa od strzała, a w nocy za cholere nie chciało, odrazu zwis ale tylko w trybie skanowania kanałów. Niby nic nie zmieniałem, nie wiem czary? :D

ErikS-web, 2024-01-07:
Disconnect the antenna and the freezing will stop.
Press M twice
Then reduce the LNA$ and/or LNA parameter.
Connect the antenna again and see if the value can be slightly higher.

zylka5, 2024-01-07:
No właśnie nie wiem czy bez anteny dao by się wejść jak problem powróci zobaczę.
Znalazłem inny mały błąd: jak odbiera się jakiś sygnał audio w spectrum to po wyjściu z niego dalej świeci się zielona dioda i słychać lekki szum w głośniku, znika to dopiero po zmianie kanału lun częstotliwości na inny.

closed 2024-01-09 by kamilsss655