kamilsss655/uv-k5-firmware-custom #47: The BUG in AM reception?

chouyu, 2024-01-08:
After removing the AM Fix code, v.17.1 can receive weak AM signals normally. However, the SLOW AGC produces a burst of noise at the beginning of strong signals, indicating insufficient dynamics. There are two issues with FAST AGC: first, low volume; second, it cannot handle extremely strong signals (e.g., signals when an airplane flies overhead), sometimes resulting in a "PUPUPUPU" sound.

chouyu, 2024-01-08:
And the sound level not normal, some signals very loud, some signals very silent. 
The issue of uneven volume is not as severe and pronounced when using the AM Fix mode.

zylka5, 2024-01-08:
Jeszcze taki mały błąd jak mamy na VFO skopiowaną cześtotliwość z kanału (F + 1) to po zmianie w menu rodzaju AGC znika skopiowana czestotliwość i wraca "stara".

kamilsss655, 2024-01-08:
Thanks @chouyu for detailed information. Based on that I've prepared new release 17.2 can you see how that works for you?

chouyu, 2024-01-09:
Version 17.2 has better performance compared to version 17.1, and most signals are received normally. However, the SLOW AGC produces a burst of noise at the beginning of strong signals, and then the AGC kicks in to adjust the signal. When encountering extremely strong signals, it still clips significantly. In terms of volume, it is still not as stable as in the AM Fix mode.

kamilsss655, 2024-01-09:
Ok, I got some other positive reports as well. Closing.

closed 2024-01-09 by kamilsss655

chouyu, 2024-01-09:
It seems that in the SLOW/FAST AGC modes, the gain is higher compared to the AM Fix mode. As a result, even slightly stronger signals may exhibit clipping distortion in the demodulated audio.