kamilsss655/uv-k5-firmware-custom #52: Switching from channel mode to VFO does not tune to the correct frequency shown

Xygen, 2024-01-09:
While listening to a transmission in channel mode, and then switching to VFO, the radio does not tune to the frequency shown in VFO mode. Instead one can still hear the transmission of the channel. One needs to manually change the frequency (like pressung up and down buttons) in order to be able to hear the true transmission of the frequency shown in VFO mode.
@kamilsss655 I've sent you 2 videos about that bug. But I can upload them here as well, if that's necessary.

sparky93a, 2024-01-09:
confirmed here also. sometimes it selected vfo once chn tx ended.

zylka5, 2024-01-09:
The same situation is after exiting spectrum mode.

xes, 2024-01-09:
Maybe this is related also with some weird behavior of spectrum view.  
Sometimes you open spectrum and you discover that it is centered on a frequency / band completely different from the one you were receiving before.

commit 910120d8 (2024-01-09): fix #52 switching from channel mode to VFO does not tune to the correct frequency shown
closed 2024-01-09 by kamilsss655