kamilsss655/uv-k5-firmware-custom #74: messenger: can't send messages in 19.4, in 19.3 worked fine

T1MOXA, 2024-01-17:
When pressing menu button to send text, radio just beeps, whereas in the old version, a message was sent

kamilsss655, 2024-01-17:
Cannot reproduce. Works fine.

Maybe it's your BW, power, conditions.

closed 2024-01-17 by kamilsss655

T1MOXA, 2024-01-17:
The difference between two radios is:
CFW 19.3 and 19.4
models, uv k5 (k6) and uv 5r plus
Maybe some settings.

T1MOXA, 2024-01-18:
Any idea how to fix this issue ?

kamilsss655, 2024-01-18:
Since you haven't shared details on your settings I was not able to reproduce the issue.
Because of that I cannot help you further.

T1MOXA, 2024-01-18:
> Since you haven't shared details on your settings I was not able to reproduce the issue. Because of that I cannot help you further.

II'd be happy to help with debugging.
I just don't know what could be the cause other than what I've already listed.

T1MOXA, 2024-01-20:
Updated CFW to v.19.8, and problem persists.
I also checked setting through menu, side by side, and all settings was identical, even frequency is the same.
I don't know what else can cause the problem. May be different versions of hardware ?

T1MOXA, 2024-01-27:
Looks like the problem was fixed in newer firmware!