kamilsss655/uv-k5-firmware-custom #77: [FEATURE] Please use a title that makes sense

KerseyPaul, 2024-01-18:
Dear Kamilsss655;

Thank you for your time and efforts in developing this great firmware!

Feature request: would it be possible for the spectrum analyzer (while operating in channel mode) to automatically switch between FM/AM/SSB as per previously set in the channel/memory slot?

Thanks in advance, best regards! 💪👏🍻👍

kamilsss655, 2024-01-18:
Hi @KerseyPaul I do have some enhancement planned that would address what you've mentioned and a couple of other things.

Not sure if it will work and if it will be implemented, but stay tuned! :)

kamilsss655, 2024-01-19:
I do have that planned :)

kamilsss655, 2024-02-04:
I was considering these changes, but based on bin size restraints and also the fact that dynamically changing rx params when we enter rx mode, will cause the rssi to change and cause glitchy squelch issues etc it is unlikely to be added.

closed 2024-02-06 by kamilsss655