kamilsss655/uv-k5-firmware-custom #82: [BUG] Volume random in AM v.19.6

chouyu, 2024-01-18:
**Details:**
- Firmware version: v.19.6
- Bandwidth: 25k
- RxAGC: FAST/SLOW
[video](https://photos.app.goo.gl/3DTMAchZM6az4JD6A)

kamilsss655, 2024-01-18:
Lot's of static - likely low signal level?
Does it behave differently with v.19.5 ? If so can you make a recording?

chouyu, 2024-01-18:
The signal -106 ~ -108 dbm on screen.
Preview run firmware v.19.3 is normal of this signal.
Didn't change anything but upgrade firmware.

kamilsss655, 2024-01-18:
Can you make a recording of the same conditions with v.19.3 ?

chouyu, 2024-01-18:
Uploaded to the same url.

kamilsss655, 2024-01-18:
Thank you, this is really helpful!

I will make some adjustments in the next release based on your feedback.

kamilsss655, 2024-01-20:
@chouyu is the problem gone with v.19.7 or v.19.8?

chouyu, 2024-01-20:
It is better. 
However, when a strong signal is incoming, it will overdrive, resulting in a loud audio with clipping, similar to the original firmware receiving AM. 
When receiving a strong signal, in my opinion, AM Fix > FAST AGC > SLOW AGC.

kamilsss655, 2024-01-20:
Ok, cool.

AM Fix adjusts the same settings you can adjust yourself in spectrum fine tuning menu.
Can you try experiment with LNAs, LNA, PGA gain settings in spectrum and see if any settings fix overdrive issue?

chouyu, 2024-01-20:
It is difficult to adjust parameters by manual testing because the received signals are in the approach phase, and each communication usually lasts no more than 10 seconds, which makes it impossible to adjust manually quickly. I will find an opportunity to record a video of the overdrive when receiving a strong signal.

kamilsss655, 2024-01-20:
You do not have to adjust settings quickly.

Just check couple of different settings (especially with LNAs 1,2,3) and listen to multiple transmissions and let me know the results.

chouyu, 2024-01-20:
In manual setting, I can't know which is overdrive signal. I only have one K5, another Air Band receiver is Wouxun uv9dp, it is better rig, have not the save issue, all signal receive normal.

chouyu, 2024-01-20:
Here is another issue, my K5 can't receive base signal if bandwidth narrow than 25khz.

chouyu, 2024-01-21:
Upgrade 2 videos in album.

kamilsss655, 2024-01-21:
Nice!
What version is that?
RxAGC = Fast?
LNAs, LNA, PGA = default?

chouyu, 2024-01-21:
v.19.8 and FAST AGC, LNA, LNA, PGA default.

chouyu, 2024-01-21:
v.19.9 clipping video uploaded.

kamilsss655, 2024-01-21:
There were no changes to VFO reception in V.19.9 vs V.19.8 so likely you are getting stronger signal now.
I'd suggest playing with the LNAs, LNA, PGA settings to find the right spot.

kamilsss655, 2024-01-21:
Ohh you can also try listening in spectrum channel mode `F+5`, as it doesn't use dynamic filter bandwidths so it might results in more clear audio. Let me know if that helps.

chouyu, 2024-01-22:
pls don't use my video in readme or wiki.
tnx.

kamilsss655, 2024-01-22:
ok