kamilsss655/uv-k5-firmware-custom #90: [BUG] DWR not working as expected

fsclavo, 2024-01-21:
As mentioned on Telegram, Dual Watch Respond isn't working as expected: even when secondary channel is on RX, if I press PTT the radio transmits on primary channel.

FW Version: v19.8

kamilsss655, 2024-01-22:
Thanks @fsclavo.

Does the issue affect @egzumer fw as well?

ErikS-web, 2024-01-22:
See Wiki

under [10 ‐ Radio operation](https://github.com/kamilsss655/uv-k5-firmware-custom/wiki/10-%E2%80%90-Radio-operation)

- The main channel is marked ► and with PTT active ►TX

_Read Main also as Primary..._

under [20 ‐ Menu](https://github.com/kamilsss655/uv-k5-firmware-custom/wiki/20-%E2%80%90-Menu)

47. `RxMode` - sets how how the upper and lower frequency is used
   * MAIN ONLY - always transmits and listens on the main frequency (NO extra characters on screen)
   * DUAL RX RESPOND - listens to both frequencies, if signal is received on the secondary frequency it locks to it for a couple of seconds so you can respond to the call (**`DWR`** on screen)
   * CROSS BAND - always transmits on the MAIN/primary and listens on the secondary frequency (**`XB`** on screen)
   * MAIN TX DUAL RX - **always transmits on the primary**, listens to both (**`DW`** on screen)

fsclavo, 2024-01-22:
> Does the issue affect @egzumer fw as well?

The last egzumer version I've used was v0.20, as far as I can remember DWR worked fine.

kamilsss655, 2024-01-22:
Yup, this bug doesn't occur on latest egzumer.

closed 2024-01-22 by kamilsss655

fsclavo, 2024-01-22:
Wow, that was fast! I could try in a few hours when I get home: will try v19.10 and if I found any problem, I'll let you know.
Thanks!

ErikS-web, 2024-01-24:
@fsclavo: Any time to test ....and what is the result ?

Solution was....
[image]

And it is included in[latest release v.19.10 ](https://github.com/kamilsss655/uv-k5-firmware-custom/releases/tag/v.19.10)

fsclavo, 2024-01-24:
I've just updated my radio to v19.10 and DWR is working as expected.
Thanks!

commit cd51908b (2024-01-22): fix #90 dual watch respond not responding on the right frequency