kamilsss655/uv-k5-firmware-custom #93: [FEATURE] Backlight: same behaviour in spectrum than in VFO/channel

fsclavo, 2024-01-23:
In spectrum mode the display backlight can be always ON or OFF, no matter if a signal is being received by the radio. This has some drawbacks: for example you have to turn backlight ON to see the display if the radio is on a dark environment, constantly consuming battery or leave it off and press a button when a signal is received.
I think that it could be great if the backlight turns automatically ON only when a signal is received, exactly the same behaviour as in VFO or channel mode. This also simplifies the spectrum UX (user doesn't need to turn backight ON or OFF) and it could also free a button in this mode.

MrHoerby, 2024-01-23:
Ich weiß nicht, das sind so 3-10 mA die da von der LED verbraucht werden!?

fsclavo, 2024-01-23:
Each led draws about 10mA, I don't know how many leds the backlight has.
But current draw isn't the main advantage I see.: the most important one is UX being the same in all radio modes

kamilsss655, 2024-01-27:
I've done some testing and used it under different conditions and I like the way it works now, so not gonna change it.

However @fsclavo I really admire how you described the issue, context and the use case. Keep it up! :)

closed 2024-01-27 by kamilsss655