kamilsss655/uv-k5-firmware-custom #94: [FEATURE] Display changes requested ti reduce confulsion and to simplify

VintageEngineer, 2024-01-24:
There are some icons on the display that are redundant and/or confusing. 

1. Delete extra (redundant) “R” and “T” in the upper left corner of the display that indicate receive and transmit. This is already indicated by the bold Rx or Tx shown by the active channel.

2. Make the scan group indicators consistent. Call them “I” and “II” on the screen in both the top left while scanning, to the right of the channel label, and in the manual. (Currently shows "1" or "2" in some places and "I" "II" elsewhere, very confusing.)

3. “><” indicates the response timer is running for “dual rx respond”. But it erroneously also shows when in “main tx dual rx”.
    
    Current implementation in each mode:
    dual rx respond: shows "DWR" and then shows "><" for 4 seconds after a transmit or receive
    main only: shows nothing
    main tx dual rx: shows "DW" and then "><" for 4 seconds after a transmit or receive
    crossband: shows "XB" all the time

The “><” indicator is redundant and confusing and should be deleted. I suggest that the two “>” indicators, one by each channel, are adequate: The  “>” indicates both that the respond timer (4 seconds) is active and also what channel will be responded to. 
If for some reason it is desired to keep the "><" symbol, it should at least be deleted from the screen when in "main tx dual rx".

ErikS-web, 2024-01-29:
Why change. Isn't it QS-default ?

kamilsss655, 2024-01-29:
Haven't look into it yet, but if some of these suggestions provide binary space reductions I will consider them.