kamilsss655/uv-k5-firmware-custom #95: [FEATURE] Include MDC1200 or dtmf feature to compile alternately in makefile.

IvanMaestroGTR, 2024-01-25:
Hi
While using this firmware, I found out the dtmf feature is missing, since I am using my radio for group calling, so sometimes I use the ptt id to id myself and my mates.

So I wonder is it possible to add the ptt id feature only to this firmware, and is it possible to edit makefile to accomodate mdc1200 feature from 1o11 firmware into this firmware?

ErikS-web, 2024-01-25:
As HAM we don't use MDC1200 as is possible in some commercial radio's. We speak out our licence call.

There is only 60 Kb of memory, so better to implement other more useful things.

kamilsss655, 2024-01-25:
There are other forks that support these functionalities.
This one is focused on different ones, and will not support DTMF or MDC1200.

closed 2024-01-25 by kamilsss655