kamilsss655/uv-k5-firmware-custom #97: Channel scanning works only on scan list 1+2

magnetra7, 2024-01-26:
Hello,

I just uploaded a channel list with CHIRP to my Quansheng UV-K5 8 with your firmware version 19.10. Some channels are listed for channel list 1, some for channel list 2 and some for channel 1 and 2. When in the channel scan mode, it only scans when I scan channel lists 1+2 (*). When I hold the scan button to switch to channel list 1 or 2, it doesn't scan all channels and seems stuck. Maybe this is due to me not using the CHIRP Egzumer module version, because I was afraid I couldn't open and flash my channel list from the normal CHIRP (non egzumer) version.

kamilsss655, 2024-01-26:
Can someone confirm this?

magnetra7, 2024-01-26:
Okay, I just did 2 things different:
I loaded my channel list into the egzumer version of CHIRP and I added some channels to channel list 1 (I had none on channel list 1). Now switching between 1, 2 and 1+2 (*) works.

ErikS-web, 2024-01-26:
You mean the SCAN started from a `memory channel` with the `*`-button ? (see [button description](https://github.com/kamilsss655/uv-k5-firmware-custom/wiki/30-%E2%80%90-Button-functions#-scan))
And switching between scnalist 1,2,ALL with the `*-button ?

### **Work fine for me**... using the EGZUMER-chirp 👍

kamilsss655, 2024-01-26:
Thanks @ErikS-web !

closed 2024-01-26 by kamilsss655

magnetra7, 2024-01-26:
> You mean the SCAN started from a `memory channel` with the `*`-button ? (see [button description](https://github.com/kamilsss655/uv-k5-firmware-custom/wiki/30-%E2%80%90-Button-functions#-scan)) And [...]

Yes.