kamilsss655/uv-k5-firmware-custom #100: [FEATURE] While switching from Memory mode to VFO mode, pressing FN+3 instead of holding 3 will take over the frequency

Xygen, 2024-01-27:
In order to have a specific frequency in VFO, which is stored in a memory channel, it's currently necessary to type the frequency.

**Describe the solution you'd like**
While in memory channel mode, pressing F+3 should switch to the frequency of the last channel.

**Additional context**
This replicates the behavior of IJV.

ErikS-web, 2024-01-27:
Works as designed and explained in the[ Online Manual under `3`](https://github.com/kamilsss655/uv-k5-firmware-custom/wiki/30-%E2%80%90-Button-functions#3-vfomr)

So returns to last used VFO-frequency on that channel.

Suggest to set menu 31. `ChDisp` - channel display style  to **Name + Freq or Number**. So you can see what frequency is that memory channel. 

What is the benefit of changing this functionality?

Xygen, 2024-01-27:
Thanks Erik.
My issue: I have the frequency of the ISS repeater stored in a channel. But in order to be able to adjust the frequency for the doppler effect, I need switch to VFO.
Currently I have to type the frequency.
I'd love to be able to press `F+3´ (also used by IJV) from a memory channel and having the frequency of the memory in VFO would make this a little more comfortable.
Thank you!

Xygen, 2024-01-27:
> So returns to last used VFO-frequency on that channel.

The part "on that channel" is not correct. Holding `3` will always switch to the last frequency used on that particular VFO.

ErikS-web, 2024-01-27:
You have 2 VFO's "in memory"

Suggest to keep VFO of channel B especially fot that use purpose

I think the radio-buttons are not easy to easy with sat-communication. I prefer a rotating KNOB.

I suggest to have a look at QS-DOCK project.

p.s. 
Directing to other project which are NOT Open Source makes no sense.
This project is, so you could change it for yourself ; home adapted  firmware;-)

kamilsss655, 2024-01-27:
@Xygen this feature is actually in the firmware, but it is undocumented in the wiki.

Long press `1` or press `F+1` while in memory mode to copy MR to VFO.

Wiki needs updating.

cc @ErikS-web,

Xygen, 2024-01-27:
Awesome! Thank you!
I can update the Wiki, if you don't mind.

kamilsss655, 2024-01-27:
I think we should ask @ErikS-web as he is the main maintainer of the Wiki.

Erik are you ok with that?

ErikS-web, 2024-01-27:
True, never used that. And was [already documented](https://github.com/kamilsss655/uv-k5-firmware-custom/wiki/30-%E2%80%90-Button-functions#1-band)

Xygen, 2024-01-27:
Awesome guys! Thank you!
BTW: The wiki can be edited publicly... Not sure if it's supposed to be.

kamilsss655, 2024-01-27:
Ohh I did't notice that. Thanks @ErikS-web !

closed 2024-01-27 by kamilsss655