kamilsss655/uv-k5-firmware-custom #106: [BUG] Messanger: Constant ERROR: INVALID PACKET flood.

T1MOXA, 2024-01-28:
**Describe the bug**
When sending a message in messenger after receiving the message, it starts flooding with errors.

**To Reproduce**
Steps to reproduce the behavior:
1. Prepare two radios on the same frequency and same encryption key.
2. Go to messenger.
3. Type message.
4. Send message.
4. See error.

**Expected behavior**
There should be no error messages.

**Pictures/video**
https://github.com/kamilsss655/uv-k5-firmware-custom/assets/19938629/c5fda6e1-f9c9-46d6-941b-a29f05830603

**Details:**
- Firmware version: v20
- Bandwidth: 12.5K
- Power: L
- RxAGC: SLOW
- Gain settings (LNAs/LNA/PGA): ?
- other: Encryption key 0123456789

kamilsss655, 2024-01-28:
Nice report!

There are 2 issues here:
1. The firmware has a bug where it ACKs the invalid packets, when it should only ACK valid ones.
2. You're receiving invalid packets because either/or:
   * your radios are too close to each other
   * your non-standard antennas affect the signal
   * your bandwidth is too narrow

T1MOXA, 2024-01-28:
2. I've tested 25k bandwidth.
Antenas have good SWR on selected frequency but I'il try stock one.
And on video I've tried move redio for about 1 meter away.

T1MOXA, 2024-01-29:
UPD.
The same problem with stock antennas, BW 25k and distance between radios about 10 meters.

kamilsss655, 2024-01-29:
Interesting, I tried reproducing it but I couldn't make it send invalid packet.

Can someone else confirm this?

MrHoerby, 2024-01-29:
Ich hatte das auch gestern bei zwei Aussendungen, aber ich hatte beide Geräte in den Händen und die Bandbreite auf 5 kHz eingestellt. Leider ist die Schrift im Messenger zu klein und ich habe dann lieber die DTMF-Sachen im Makefile aktiviert.

T1MOXA, 2024-01-29:
Looks like I found the cause.
It's in ROGER = MDC (39 Parameter in the menu)
With ROGER = Roger || ROGER = OFF, error gone.

kamilsss655, 2024-01-29:
Awesome! I am glad you figured it out.

kamilsss655, 2024-01-29:
I see you submitted a PR. Awesome! :tada: 

I will review it tomorrow.

commit 12263a76 (2024-01-30): Fixes #106
commit 9c6bfab2 (2024-01-30): Merge pull request #109 from T1MOXA/fix/messenger-roger-error

Fixes #106