kamilsss655/uv-k5-firmware-custom #111: [BUG] When connected to a repeater, the morse code beep enables the messenger read reciept tx.

IvanMaestroGTR, 2024-01-30:
Whenever the nearest repeater transmits the id morse code the uvk5 misread the signal as message and tx read reciept.

I suppose is it possible to disable messenger in some channel?

kamilsss655, 2024-01-30:
Confirmed. Possible improvement is on the roadmap.

IvanMaestroGTR, 2024-01-30:
Oh btw I noticed when you send messages, there will be a short period of time where the mic is activated

Is it possible to make it muted?  Or maybe insert a universal code (for every k5 with this mod), so that the messenger will not be activated too easily?

T1MOXA, 2024-01-30:
> The mic activation has been reported here: #68 and should be fixed in v.20. Can you re-check?

Yes, I noticed this problem in v.20 too

kamilsss655, 2024-01-30:
Confirmed as well. I have re-opened the #68.

IvanMaestroGTR, 2024-02-02:
From my s meter it shows +10, -98 db

commit 15e012a3 (2024-02-02): fix #111 false positive message rx
closed 2024-02-02 by kamilsss655