kamilsss655/uv-k5-firmware-custom #115: [BUG] Extended VFO shows a full bar even the signal very faint

Xygen, 2024-01-31:
**Describe the bug**
When entering Extended VFO from channel spectrum a very faint signal is already showing a full bar.

**Expected behavior**
I'd expect a bar which is filled according to it's dBm value.

**Pictures/video**
[image]

**Details:**
- Firmware version: 20.1
- Bandwidth: 12.5k
- Power: High
- RxAGC: Fast
- Gain settings (LNAs/LNA/PGA): 3/5/6
- other:

**Additional context**
I can't reproduce this right now, but Kamil asked me to file an issue including the picture.

Xygen, 2024-01-31:
https://github.com/kamilsss655/uv-k5-firmware-custom/assets/4473704/7dabca71-e15b-490c-9dd0-ffd183748b4d

Here's the video for this issue

Xygen, 2024-01-31:
More info on the version where this issue has been introduced: With v19.10 this bug did not exist. It seems to be introduced with v20

kamilsss655, 2024-02-04:
Closing this one, @Xygen let me know if you manage to reproduce it reliably and provide the steps needed to reproduce it.

closed 2024-02-04 by kamilsss655