kamilsss655/uv-k5-firmware-custom #116: [BUG] Triggering the 1750 Hz tone by pressing PTT+Fn2 also triggers scanning

Xygen, 2024-01-31:
**Describe the bug**
Some repeaters need to be "opened" by sending a 1750 Hz tone. To do this you press PTT and the Fn2 key afterwards.
The bug is that the unit is starting a frequency or channel scan, which is something the users normaly not wants to do in this scenario.

**To Reproduce**
Steps to reproduce the behavior:
1. Turn the device on
2. Tune to a frequency or a memory channel which allows TX
3. Press PTT and Fn2 in order to send a 1750 Hz tone
4. A scan is started which is not expected

**Expected behavior**
As a 1750 Hz tone is used to open the repeater, it is expected that the device stays on the currect memory channel or frequency.

**Pictures/video**
A video can be provided if necessary.

**Details:**
- Firmware version: 20.1
- Bandwidth: any
- Power: any
- RxAGC: any
- Gain settings (LNAs/LNA/PGA): any
- other: -

**Additional context**
There is a info that this bug has not existed in v19.1. If necessary, we can test which version introduced the bug.
Also it seems not every device is affected. But at least 2 known users are affected.

Xygen, 2024-01-31:
I have to add, that when sending DTMF tones while pressing PTT also triggers the standard function of the key, when it's being pressed a little longer. For example Pressing the number 6 for half a second (while pressing PTT) also switches the output power.
These function shouldn't be executed while pressing PTT.
Thanks guys!

Xygen, 2024-01-31:
https://github.com/kamilsss655/uv-k5-firmware-custom/assets/4473704/a4e9bbac-54a3-40a0-a3bb-f08f81320a12

Here's the video which show's the issue in VFO mode. The same happens also in memory mode.

Xygen, 2024-01-31:
More info on the version where this issue has been introduced:
With v19.10 this bug did not exist. It seems to be introduced with v20

cetebe, 2024-02-02:
since v20 i have a similr problem on a k6 and R5, while pressing PTT and another key, like A, B , C, D, the action of that key will happen, instead of just sending the tone.
For example, holding PTT + press channel up stops immediately transmitting and perform the channel up action! total crazy behavior. With 19.10 everything was fine...

commit f1243430 (2024-02-02): fix #116 Triggering the 1750 Hz tone by pressing PTT+Fn2 also triggers actions
closed 2024-02-02 by kamilsss655