kamilsss655/uv-k5-firmware-custom #118: [BUG] Since 20.2 audio stops randomly

Xygen, 2024-01-31:
**Describe the bug**
Since 20.2 while listening to a transmission on 70cm (haven't tested other bands), audio stops randomly, while the signal indication (dBm value and green light) persists. To get the audio back one must change the channel up and down, or change a setting like bandwidth or TX power.

**To Reproduce**
Steps to reproduce the behavior:
1. Listen to a conversation on 70cm (haven't tested other bands) in any demodulation mode.
2. Put SQL to 0 to make sure it's not squelch closing
3. After a few seconds the audio stops, 
4. To get the audio back, one needs to change a setting, like bandwidth or output power. But after a few seconds the audio is gone again

**Expected behavior**
Audio shoudn't drop out during RX

**Pictures/video**
If applicable, add pictures or video to help explain your problem.
To insert a picture (.jpg), you can simply drag it to this window.

**Details:**
- Firmware version: 20.2
- Bandwidth: any
- Power: any
- RxAGC: any
- Gain settings (LNAs/LNA/PGA): default
- other: -

**Additional context**
I wanted to provide a video but struggeling with the 10MB limit...

kamilsss655, 2024-01-31:
Confirmed. Priority: high.

kamilsss655, 2024-02-01:
In 20.2 I added mute after message received interrupt is triggered, so the user doesn't have to listen to FSK modulation of incoming message. It seems it is triggered randomly during regular RX causing this issue.

In next version 20.2.1 I am moving it to only mute if the message sync word is received. If it causes further issues, the mute will be completely removed.

kamilsss655, 2024-02-02:
With `v.20.4` this issue still persists although it is much more rare.

It happens during RX of channels that use CTCSS.
I will be adding condition to only allow mute if no CTCSS channel is used, if that doesn't solve it then this feature will be deactivated to prevent mutes.

kamilsss655, 2024-02-04:
With `v.20.5` users can also disable it with `ENABLE_MESSENGER_FSK_MUTE` Makefile flag.

kamilsss655, 2024-02-05:
Closing for now, please report here if you experience this issue.

closed 2024-02-05 by kamilsss655

silasb, 2024-04-16:
I have this same issue with `v.20.5`.  Channel 1, 2m 147.110Mhz, -0.6Mhz offset, CTCS 67.0Hz. If I manually open squelch I can hear just fine.

silasb, 2024-04-16:
I've tested with `MsgRx` off.  I've also compiled a firmware with:

```
ENABLE_MESSENGER = 0
ENABLE_MESSENGER_FSK_MUTE = 0
```

With CTCSS it works perfectly.

Video - https://github.com/kamilsss655/uv-k5-firmware-custom/assets/40670/1c313328-d3e2-4dc0-b621-7c216f19274a

silasb, 2024-04-27:
Making more progress on this.  This only seems to occur on RXCTCS 67.0Hz on the frequency listed above.  I don't have any other repeaters near me that use 67Hz so I can't tell if it's a frequency issues as well.

commit 4bf3da5e (2024-02-01): Merge pull request #119 from kamilsss655/rc20.2.1

fix #118 audio stops randomly
commit 7ca2015d (2024-02-01): fix #118 audio stops randomly