kamilsss655/uv-k5-firmware-custom #124: FR: import Egzumer's latest F/W AM Demodulation settings

KerseyPaul, 2024-02-03:
Many UV-K5 users (myself included) on dedicated Facebook groups are reporting a considerable improvement on the airband RX audio quality (AM demodulation) since Egzumer's latest v0.22 firmware. 

Would it be possible to import the related modifications in order to achieve the same improvement? 

Thanks in advance, best regards and keep up the great work! 💪🍻

FrostyMisa, 2024-02-04:
I'm not sure. I do test with my two Quasheng radio, one with Egzumer firmware and one with this one, and I find that they have identical reception when placed side by side. I live 11 km away from the airport, where I listen to their ATIS, and I have a better experience with this firmware because Egzumer sometimes produces an annoying "repeated clicking" sound, which is not present in this one. Yes, both display different signal strength on the S meter, but even when they show different strengths, they lose signals under similar conditions.

kamilsss655, 2024-02-04:
Thanks @FrostyMisa this is also my experience.

@egzumer and I use different approaches to AM demodulation which is great because users can choose what works best for them.

closed 2024-02-04 by kamilsss655