kamilsss655/uv-k5-firmware-custom #131: [BUG] Unable to set TxPwr to OFF on Air band after set LOW, MID or HIGH

batona, 2024-02-05:
Unable to set TxPwr to OFF on Air band after set LOW, MID or HIGH.
Actually unable set TxPwr to OFF on any other band and mode. No OFF option at all.

**To Reproduce**
Steps to reproduce the behavior:
1) set VFO to air band for example 122000
2) Press menu and navigate to TxPwr (set to OFF as default)
3) Press menu again and set TxPwr to LOW or any other
4) Comfirm choice
5) Attempt to change TxPwr to OFF - no such option when pressing UP-DOWN keys. Only LOW, MID, High

**Expected behavior**
OFF - LOW - MID - HIGH options on TxPwr menu item

**Details:**
- Firmware version: NUNU 20.5
- Bandwidth: 25k
- Power: OFF
- RxAGC: LSOW
- Gain settings (LNAs/LNA/PGA): 
- other:

**Additional context**

kamilsss655, 2024-02-05:
Yea I've noticed that power setting sometimes is not displayed. Not sure how that's supposed to work.

I will look into it.

kamilsss655, 2024-02-12:
Looking at the code, there should be no `OFF` option, so the bug is rather that something sets `TxPwr` to `OFF`.
I've might have experienced it once, however I was not able to reproduce it.

If someone manages to reproduce it, please post the steps here.

closed 2024-02-12 by kamilsss655