kamilsss655/uv-k5-firmware-custom #133: [BUG] Messenger ACK not sent on secondary frequencies [A/B]

tonurics, 2024-02-05:
Settings:
- RxMode = MAIN TX DUAL RX
- MsgAck = True

Firmware 20.5: when receiving a message on the secondary frequency [B] an ACK is not sent back to the sender.

I suspect the ACK is being sent out on the primary frequency [A]. But I have not confirmed this.

kamilsss655, 2024-02-05:
Well of course it responds on the main frequency, because that's what `MAIN TX DUAL RX` is supposed to do.
If you switch to `DUAL RX RESPOND` it will ACK on the right frequency. :)

closed 2024-02-05 by kamilsss655

tonurics, 2024-02-05:
Yes, I understand that would work...

But since the ACK is automatic: I was hoping it would behave like DTMF REPLAY. Where a DTMF is sent back to the caller; regardless of RxMode. (At least that has been my experience.)