kamilsss655/uv-k5-firmware-custom #136: [FEATURE] Lower binary size

kamilsss655, 2024-02-06:
Currently we're nearing the maximum allowed binary size for this radio. It would be great to find out the way to lower the binary size further. There might be some more research needed into methods allowing for further size reduction.

**Describe the solution you'd like**
To find an actionable way of reducing the binary size of the images.

Any help would be appreciated.

ElDiablo88, 2024-02-07:
maybe could be used docker on local machine and see if it will compile the bin with less memory usage?

repocent, 2024-02-07:
If taken less size; made one without messenger and with DTMF back? Completely stripout FM radio?

kamilsss655, 2024-02-08:
I think there is a way to do profiling/instrumentation optimizations by using SWD connection.
[https://mcuoneclipse.com/2015/08/23/tutorial-using-gnu-profiling-gprof-with-arm-cortex-m/](url)

zhangyulu8023, 2024-02-10:
FM is unuseable,please move it out

kamilsss655, 2024-02-10:
Just to be clear for those who didn't understand the issue subject. The issue about lowering the binary size, not disabling the `Makefile` flags which everyone can do based on their preference and needs.

jasperchua99, 2024-02-11:
I think some of the defination that has been defined could be move stright to the ans to save a few bytes?

Like for example 

Let i = 1
Return i + 2

Change to 

Return 1 + 2

dchauran, 2024-02-14:
Looking through the features on the radio, I din't think I have ever used scrambler or compander, so perhaps making those flags and disabled by default?

kamilsss655, 2024-02-14:
`settings.c` and `board.c` can be size reduced by refactoring EEPROM map to be a single union structure with serialized array representation. Then settings savings and loading can be a single for loop.

I'd take a PR, is someone wants to look at it.

Lar-Sen, 2024-02-18:
There are some pieces of code which don't have real use, e.g the "band" thing : The device could be set to have rx limits all over BK4819 capabilities, and tx limits restricted to a simpler "band-pass". Anyhow it was never meant to gain legality validation (+ we have an encrypted messenger here, which is de facto illegal in many countries). It's the owner's responsability to wisely use frequencies.
The actual 8 band implementation takes some bytes and serves nothing in my opinion.

closed 2024-04-12 by kamilsss655