kamilsss655/uv-k5-firmware-custom #140: Spectrum scanner stopped working on memory mode

fsclavo, 2024-02-12:
I've been using my radio daily without issues but today, when I try to start spectrum scanner, it stopped working: the showed frequency is 0.00000 and the radio completely hangs. Only powering it off brings it back to life. On VFO mode the spectrum scanner works perfect.
Until today, I've used the scanner dozens of times without issues.

[image]

kamilsss655, 2024-02-12:
This is invalid issue. Please use a template next time.

As a courtesy I should mention that I've seen this on several occasions. This is some rare bug that is hard to reproduce. Likely will not be fixed.

closed 2024-02-12 by kamilsss655

fsclavo, 2024-02-12:
You're right @kamilsss655 , will use template next time, sorry.
Just FYI (maybe useful to diagnose in the future) it happens much more often if you turn on the radio and starts spectrum quickly. FW version is 20.4.