kamilsss655/uv-k5-firmware-custom #142: [BUG] Messenger security: Leaked encryption key

Lar-Sen, 2024-02-12:
**Describe the bug**
Messenger encryption key is cleartext readable everytime.

**To Reproduce**
1. Set encrytion key
2. Dump EEPROM using any tool requesting UART 051B command, e.g. https://github.com/Lar-Sen/Quansheng_UV-K5_Kitchen/blob/main/v2.01.31_mods/utils/eeprom_tool.py
3. Read encryption key at 0x0F30
4. Dun goofed.

**Expected behavior**
Genuine firmware uses EEPROM 0x0F30 offset to store hashed config password which is used as an AES key. If config password was not given since last boot ('login'), the UART command 0x51B just fails, which is expected behaviour.
In our case, as 'config password' routines were circomvented, EEPROM offset 0x0F30 should NEVER get dumped.

Fix: UART routine must send 0xFF in place of key data if block 0x0F30-0x0F3F gets requested (or just restore original 'login' process before allowing dump).

**Details:**
- Firmware version: RC20.5
- other: UV-K5(8) HW rev. 1.6

**Additional context**
In case of a group scenario, users must share the same encryption key.
Original firmware didn't allow 0x0F30 eeprom block to be restored. In our case, I think it's better to allow it, provided that block couldn't get dumped in any way.

...anyway, this fork is still the best in my opinion :)

kamilsss655, 2024-02-12:
Well you are wrong.

Yess, the EEPROM password is stored in plaintext, which is clearly mentioned in the [Wiki](https://github.com/kamilsss655/uv-k5-firmware-custom/wiki/44-%E2%80%90-Encryption):

> Of course if the EEPROM password is read, it can be copied to other devices and these devices will derive it to the same encryption key, which is actually desired as it allows setting up multiple encrypted radios in bulk via CHIRP.

The solution you propose doesn't provide any additional security, considering the EEPROM itself is not encrypted, users could flash other firmware that would still allow for EEPROM dump.

What you propose (locking EEPROM read, without FW re-flashing) can already be achieved by setting the `Passwd` option in menu. This way serial connection is prevented until correct password is entered on boot.

closed 2024-02-12 by kamilsss655

kamilsss655, 2024-02-12:
It is a good exercise in terms of security though. :)

Based on your issue I've enabled `Private vulnerability reporting` for this repository.

kamilsss655, 2024-02-12:
> edit: I don't think this case should be kept private :) You decide.

I guess you are right. Let's keep it public, to prevent duplicates.

kamilsss655, 2024-02-12:
Oh and I don't like that approach with AES key hashing.

The only proper hardening solution in my opinion would be to encrypt EEPROM with `Passwd` derived key. The EEPROM would have to be decrypted on each boot based on user entered key. After 3 wrong retries the EEPROM would get reset.

This would take too much space of the binary image though, so I will not be implementing that.