kamilsss655/uv-k5-firmware-custom #153: [FEATURE] F LOCK PMR LPD

merc07, 2024-02-19:
I see you have added PMR446 to the F Lock menu, good idea
Could we add LPD, either in addition with PMR as below or keep just your PMR and add option of PMR LPD?

case F_LOCK_PMR:
			if ((Frequency >= 44600625 && Frequency <= 44619375) || (Frequency >= 43307500 && Frequency <= 43477500))
				return 0;
			break;

const char * const gSubMenu_F_LOCK[] =
{
	"DEFAULT+\n137-174\n400-470",
	"FCC HAM\n144-148\n420-450",
	"CE HAM\n144-146\n430-440",
	"GB HAM\n144-148\n430-440",
	"137-174\n400-430",
	"137-174\n400-438",
	"PMR LPD",
	"DISABLE\nALL"
};

kamilsss655, 2024-02-19:
I am not planning on adding that.
Note: delta between lowest power of this device (~0.4W) and the max allowed power at LPD (10mW).

kamilsss655, 2024-02-19:
Not to mention potential interference to remote key-less entry systems.

closed 2024-02-19 by kamilsss655

merc07, 2024-02-19:
No worries - I thought the lowest power was < 20mw, that is according to some of the firmwares such as F4HWN.
I have not tested it myself yet.

I suppose on that note power should  be automatically locked to 10mw / 500mw if LPD / PMR was selected in the F Lock but are different power levels even possible within the same band - probably not or at least not easily..

rjdverbeek-tud, 2024-05-11:
The PMR466 F_LOCK is available in the source code, but it is not present in the release 20.5. Is that correct?

merc07, 2024-05-11:
I added it for myself to the excellent work from kamilsss655 - I guess I was allowed to ?
I also set the display to show the power, set to 500mw for PMR and 10mw for LPD
As kamilsss655 rightly mentioned, if using LPD, make sure the frequency used is not either close or conflicting with ham bands...