kamilsss655/uv-k5-firmware-custom #157: [BUG] **cc1.exe: error executing win_make.bat

OE1MWW, 2024-02-24:
I tried to compile local by retrieving the source via 'git clone' and did another test by downloading the sourcs as a .zip
If I execute the 'win_make.bat' the following error comes up:

**cc1.exe: error: argument to '-O' should be a non-negative integer, 'g', 's' or 'fast'
make: *** [init.o] Error 1**

Cloning and compiling the egzumer sources on the same PC ends with no errors and results in a firmware.packed.bin of egzumer
Looks like, my compiler installation seems to be OK. Because of my lack of experience in this area, i am unfortunately stuck.

How do I handle that error/bug ?

OE1MWW, 2024-02-24:
the solution - and the reason: Makefile, line 229, has the following statement:

CFLAGS += -Oz -Wall -Werror -mcpu=cortex-m0 -fno-delete-null-pointer-checks -std=c11 -MMD

the cc1.exe compiler does not like the flag '-Oz' ;-) The 'z' seems to be an invalid option, at least for my compiler.
As I set some custom mods to '0', and used less memory, I could use the option 's' (so instead of '-Oz' now '-Os') 
and there where no more errors during compile.

kamilsss655, 2024-02-24:
Glad you figured it out yourself.

The Readme section has up-to date instructions on building this firmware.

closed 2024-02-24 by kamilsss655