kamilsss655/uv-k5-firmware-custom #158: Future release schedule plans?

tonurics, 2024-02-24:
I've noticed you started making new releases exclusive to your Patreon supporters and documented some of those releases here. I am glad to see you have a growing following on Patreon. But was curious what your future plans were for this GitHub repo.

Having done crowdfunding in the past, our team struggled with figuring the best ways come up with "added value" for our supporters. "Early Access" seems like an obvious option; but it's really a double-edged sword: software improves through iteration, which comes from testing and feedback.

In the commercial software model that role is normally filled by a paid quality assurance team. With open source the user-base does everything. Hopefully some finds your project interesting enough and has the technical wherewithal to provide useful feedback or fixes. The bigger the user-base, the better the feedback. I assume you already see where I was going with this...

Actual work-place conversation: "They want to pay extra to play the buggy, broken, unfinished version of our game? ...There really is a sucker born every minute." :rofl:

If you are struggling with "added value", I personally think the best option is to just: allow your supporters a voice. Vote on features, where people want you spending time, big project decisions, etc. Just don't be afraid to make changes after the fact if something doesn't work out.

In any case, thanks for the work you've done with your fork.

kamilsss655, 2024-02-24:
The project is doing just fine, so no need to be concerned about it's future.

I don't think of my Patreon supporters as suckers, in fact they enable me working on this project.

And if they find value in my work, it makes me happy.

I assume you are not one of these people, so not sure what you are trying to achieve here?

Take care :)

closed 2024-02-24 by kamilsss655

tonurics, 2024-02-24:
Let me reiterate with additional clarification:

I was inquiring about "what your future plans were for this GitHub repo". As you've:
- Created exclusive releases for your Patreons.
- Updated the documentation here for those releases [which is how I learned of them].
- Not updated the code base here, with the changes from those releases.

By all appearances you've taken the development of your fork private. That is perfectly within your rights to do. And more power to you, if that is in-fact what you have done. Since there haven't been any announcements around this, I asked "future release schedule plans?"

While I am not currently one of your Patreon supporters, I was not being factitious when I said I was "glad to see you have a growing following on Patreon". You need to reread what I wrote, if you think I was trying to imply that your Patreon supporters are suckers.

Beyond getting an answer to my question. I was hoping to encourage you to resume development in the open, as to facilitate collaboration. By pointing out the inherit conflict between "Early Access" and open source; there are benefits to be had from the latter, but are lost due to the conflict.

Toodles :smile: