kamilsss655/uv-k5-firmware-custom #159: [BUG] Long-Press-7 (VOX) gives ScanRng & F-NOAA gives NO action

ErikS-web, 2024-02-24:
In release 21.1 VOX is not enabled bij default (disabled via Makefile flag).

F-5 / Long press 5 gives **no action**  as [decribed here](https://github.com/kamilsss655/uv-k5-firmware-custom/wiki/30-%E2%80%90-Button-functions#5-noaa)

Long-Press-7 (VOX) now gives **ScanRng**

kamilsss655, 2024-02-26:
Confirmed. Fixed in main.
Will be available in the upcoming releases.

kamilsss655, 2024-04-13:
@ErikS-web can you check if this is now resolved with v.21.4 ?

ErikS-web, 2024-04-13:
You repaired the VOX-function in Channel - AND Frequency mode 👍 

And this works

F-5 / Long press 5 only work as [decribed here](https://github.com/kamilsss655/uv-k5-firmware-custom/wiki/30-%E2%80%90-Button-functions#5-noaa) 👍

kamilsss655, 2024-04-13:
Great, thanks Erik!

closed 2024-04-13 by kamilsss655