kamilsss655/uv-k5-firmware-custom #162: [BUG] Please use a title that makes sense

nikant, 2024-03-02:
disabling 
ENABLE_SPECTRUM_CHANNEL_SCAN            := 1

with 0 produces the following compilation error:

> In file included from app/action.c:44:
app/spectrum.h:169:6: error: #elif with no expression
  169 | #elif
      |      ^
app/action.c: In function 'ACTION_RunSpectrum':
app/action.c:264:14: error: #elif with no expression
  264 |         #elif
      |              ^
make: *** [Makefile:467: app/action.o] Error 1

IvanMaestroGTR, 2024-03-02:
Usually in order to remove this error,
You can remove certain codes that are shown from the error.
As you disable them, some codes that correlate to the disabled function will lose its context, causing this to happen.

kamilsss655, 2024-03-02:
This flag is currently required.

closed 2024-03-02 by kamilsss655