kamilsss655/uv-k5-firmware-custom #167: DTMF indicator is overlapping bandwidth used by transceiver (Bandw in menu)

krotoniec, 2024-03-08:
**Describe the bug**
When DTMF decoding is on (D Decd ON), its indicator is overlapping bandwidth used by transceiver (Bandw in menu).

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'MENU > 50 (D Decd) and turn it ON'.
2. Go back to the main screen either in frequency or channel mode.
3. See the DTMF indicator is overlapping Bandw value.

**Expected behavior**
I'm open for suggestions, but, maybe instead of using "DTMF" simple "D" would be enough, and this way there's enough room on the screen. Anyway, at the moment doesn't look great.

[image]

**Details:**
- Firmware version: NUNU c13a667
- Bandwidth: 8.33k
- Power: H

kamilsss655, 2024-03-08:
I cannot reproduce this issue.

Also:
* There is no "D Decd" option in this firmware
* c13a667 is not a valid version

Can anyone else confirm or deny this issue?
cc @ErikS-web

krotoniec, 2024-03-08:
I'm little confused. The last commit from 5 days ago is c13a667. It was compiled using custom Makefile with ENABLE_DTMF_CALLING enabled, that's when it shows in menu. It's set to disabled in your firmware by default.

[image]

[image]

It easily reproducible, I tried it on 3 different devices.
And "D Decd" it is there, that's the described DTMF calling feature, enabled with ENABLE_DTMF_CALLING.

kamilsss655, 2024-03-08:
Ohh that explains it. DTMF is not supported by this firmware, which is stated in the wiki.

closed 2024-03-08 by kamilsss655

krotoniec, 2024-03-08:
I'm sorry it's not explained in the wiki.  There is a number of contradicting statements regarding DTMF. On one hand you are explaining several of DTMF options in the menu, and then at the end you are stating - "DTMF is not supported". 
It's supported or not?

ErikS-web, 2024-03-12:
I do have to correct and say you did not read the Wiki/Online manual correctly

[image]

The specific DTMF-functions are summarised under section **REMOVED**

commit 2b24ebfc (2023-12-23): FIX #167: F7+ band accessible even if WIDE_RX = 0

(cherry picked from commit a4cefe340198443086aa714eef58995c8b9aeb13)