kamilsss655/uv-k5-firmware-custom #169: next version (?) [FEATURE]

OE1MWW, 2024-03-08:
@kamilsss655 First of all, thank you for the great firmware. I have equipped several friends (licensed radio amateurs) with the v.20.5. They are very enthusiastic about the functions, especially the 'channel spectrum scan' is unique in this form. Meshtastic, messaging and encryption are very interresting and unique too, but forbidden to be used by licensed Radio Amateurs in our country. So, the question is, is there anything else planned for the successor to v.20.5 or is this 'the final version'? I ask the question without demands or anything like that - because it is your software and therefore you make the rules.

kamilsss655, 2024-03-09:
Thank you for your message and for sharing your positive experience with my firmware. I'm glad to hear that you have equipped your friends, licensed radio amateurs, with version 20.5 and that they are enjoying the unique features, especially the 'channel spectrum scan.'

I understand your curiosity about the future development of my firmware. While I cannot provide specific details at this moment, I can confirm that there are ongoing improvements being made to the firmware available to my Patreon members. These updates go beyond just mesh networking and aim to enhance various aspects of the software.

I appreciate your understanding of the restrictions regarding certain features for licensed Radio Amateurs in your country. Please know that we I am continuously working to enhance the functionality and user experience of my firmware to meet the diverse needs of my users.

Thank you for your support and for reaching out with your question. If you have any further inquiries or feedback, feel free to let me know. Your input is valuable to me as I strive to provide the best possible experience for my users.

Warm regards.
Nunu

closed 2024-03-09 by kamilsss655

jasperchua99, 2024-03-10:
Hi, for those who are not patreon members, then when will the next version be released?

OE1MWW, 2024-03-10:
@jasperchua99  His software - his rules. ultimate questions usually lead to frustration and abandonment ;-(

jasperchua99, 2024-03-10:
@OE1MWW i agree but, i do hope there's a roadmap, especially for those who can't afford a monthly subscription as a paywall is what deters people from coming back, especially for those who don't wanna pay

OE1MWW, 2024-03-10:
@jasperchua99 Maybe someone made the mistake and mentioned the next version v.21.  The firmware is written by a freelancer / hobbyist supported by a few patreonists. So, whenever he is in the mood and his time allows it, he will be with the project. Therefore, how could there be a 'roadmap' for this ?

by the way: i had a similar project, but after i published it, there were demands - ultimate demands - ('when will you add this feature...?' 'when will you change this?', ) so i stopped publishing it again.

tonurics, 2024-03-12:
I asked about this previously: https://github.com/kamilsss655/uv-k5-firmware-custom/issues/158

OE1MWW, 2024-03-13:
@kamilsss655 

Dear Nunu,

Thank you for your prompt and informative response. I greatly appreciate the ongoing improvements you're making to the firmware, particularly for your Patreon members. It's reassuring to know that you're dedicated to enhancing various aspects of the software beyond mesh networking.

I completely understand the necessity of adhering to licensing restrictions, and I'm grateful for your efforts to accommodate the diverse needs of users like myself.

Regarding the 'channel spectrum scan' feature, I wanted to emphasize its significance under certain circumstances. While I understand the limitations, I believe this feature is incredibly valuable, especially for specialized applications.

Thank you once again for your dedication to refining the firmware and for your responsiveness to user feedback. Please do keep me updated on any developments, and if there's any way I can assist or provide further input, I'd be more than happy to do so.

Warm regards,
OE1MWW