kamilsss655/uv-k5-firmware-custom #172: [BUG] - This firmware has disintegrated the ability to listen to any frequency with my Quansheng.

IsTk0, 2024-03-11:
Hi.
I'm sorry to have to say this to someone who has put in the effort to add/delete features compared to the egzumer version. but come on, I believed in you and your firmware, after all it comes from a genuine firmware. but apparently this is not the case.

After loading your firmware into my radio, there were already errors such as signals of a power equal to 32k dbm, there are two things, either they tried to kill me or there is some problem.

Another problem is that you have to shout at the settings to be able to change and confirm. they literally need to be set 3 times to actually see the real expected changes.

Bigger problem is that I am now DEAF on the air band. I no longer hear anything even though I have VOR and TWR and everything 5 km from my house, whereas before with the default software (I know it's a terrible for this range frequency) I obviously (before) heard everything super clearly, and it really bothers me, since I have wanted to try to put your firmware to have the spectrogram and sequential scan. what really makes me angry is that I bought this UV-K5(8) just for that narrow area.

Now I think I'll conclude with my complaint via issue. nothing, because now I have to buy another radio. but please if you are aware of your firmware let everyone know before actually downloading and uploading the firmware. write it nice and big everywhere and make it clear that it is unstable. I'm sorry you had to soak my anger but understand me a little, I'm forced to buy a new radio after only 5 days it arrived after I waited 3 weeks to receive it on come on.

I hope you don't get angry and that you manage to solve the problems mentioned above for others, as you understand well that throwing away 40 bucks for another radio is a bit annoying. I am sorry. I hope your firmware can make great strides. 

Thanks for your understanding

kamilsss655, 2024-03-11:
Thank you for sharing your experience with the firmware. I understand your frustration and apologize for any inconvenience you have faced. It is important to note that there is a warning on the main page indicating that the firmware may potentially brick your radio.

Based on the issues you have described, it is possible that the problems you are encountering are related to a hardware issue rather than the firmware itself. It is recommended to thoroughly check the hardware components of your radio to ensure everything is functioning correctly.

I appreciate your feedback and will take it into consideration for future improvements. If you have any further concerns or questions, please feel free to reach out. Thank you for your understanding.

closed 2024-03-11 by kamilsss655

OE1MWW, 2024-03-13:
@IsTk0   Thank you for reaching out and sharing your experience with the firmware. I am sorry for the inconvenience and frustration you've encountered with the issues you've mentioned.

I understand your concerns and would like to address them promptly. Firstly, could you please confirm whether you saved your calibration data before loading the firmware? Additionally, have you attempted to restore this data to see if it resolves the signal and settings issues you've encountered?

Your feedback is valuable in helping us improve the firmware for all users. We will take your suggestions into consideration and ensure clearer communication regarding the firmware's stability to prevent similar experiences in the future.