kamilsss655/uv-k5-firmware-custom #173: [BUG] Serial baudrate is too high

ErikS-web, 2024-03-19:
**BEFORE YOU CREATE A BUG REPORT MAKE SURE YOU READ THIS**:
https://github.com/kamilsss655/uv-k5-firmware-custom/wiki/80-%E2%80%90-Tips-&-Tricks-and-Known-Issues

**Describe the bug**
Functional serial baudrate is out of spec's for my USB-to serial interface
Tested with CH340 and a Prolific

**To Reproduce**
Loading firmware (with bootloader) had his own speed. SoO no problem there,
Radio functions OK.
But the **_serial communication_** at this new high speed between radio and PC does not work

Tested with Chirp in Developement mede and modified PY-file.; Lookup "baud" and change 384000 to 250000

**Expected behavior**
release notes mention
- messenger uart enabled (use 250000 baud)
- CHIRP driver needs to be changed to 250000 baud as well

**Pictures/video**
[image]

**Details:**
- Firmware version: 21.2 pre-release
- Other project had same experince and explained...
https://arduino.stackexchange.com/questions/41856/ch340-max-serial-speed

CHG340 supports common baud rates: 50, 75, 100, 110, 134.5, 150, 300, 600, 900, 1200, 1800, 2400, 3600, 4800, 9600, 14400, 19200, 28800, 33600, 38400, 56000, 57600, 76800, 115200, 128000, 153600, 230400, 460800, 921600, 1500000, 2000000 baud.
250 000 baud isn't supported by that IC.
The Baud steps is either **200% or 150% of the previous value** (There is exceptions).

Also here
https://arduino.stackexchange.com/questions/32035/what-usb-to-uart-chip-provides-highest-data-rates-with-arduino-boards

- [FT232R](http://www.ftdichip.com/Support/Documents/DataSheets/ICs/DS_FT232R.pdf) (FTDI): 183 to 3M baud, including 250k, 500k and 1M, 128 RX / 256 TX FIFO
- [PL2303](http://www.electronicaestudio.com/docs/PL2303.pdf) (Prolific): 75 to 1.2M baud, not including 250k, 500k and 1M, 256 RX / 256 TX FIFO
- [CP2102](https://www.silabs.com/Support%20Documents/TechnicalDocs/CP2102-9.pdf) (Silicon Labs): 300 to 921600 baud, including 250k and 500k, 576 RX / 640 TX FIFO
- [CH340G](http://cdn2.boxtec.ch/pub/diverse/ch340g-datasheet.pdf) (WCH): 50 to 2M baud, not including 250k, 500k and 1M, unknown FIFO size

Suggest to use 230400

fenton67, 2024-03-29:
I don't know if it will be possible, but I have this dream of using messenger with the cardkb for messenger text entry.  The reason I mention is because it supports only 115,200.  Just throwing that out there cause it's always a challenge to type long messages or extended conversation.  I think it is doable: https://github.com/jon7sky/CardKB_UART.

ErikS-web, 2024-03-30:
This is about serial communication betwwn trhe Radio and the ESPRI-project.

We are experiencing what is possible as fast option. So 115K is much to low,

kamilsss655, 2024-04-13:
Resolved in v.21.4.

Works with:
 ftdi, ch340 and prolific

closed 2024-04-13 by kamilsss655