kamilsss655/uv-k5-firmware-custom #177: Header short read in CHIRP

mattfox27, 2024-04-11:
SO no matter what I do I get an error in CHIRP saying "Header short read" I have tried with the chirp driver, with your version of chirp with chirp next and no matter what I get that error.

kamilsss655, 2024-04-11:
Sorry to hear that. I wish I could help you, but I can't since you didn't provide required information for the bug issue. Not even a version number.

closed 2024-04-11 by kamilsss655

mattfox27, 2024-04-11:
Oh Sorry, it was the latest version from KoFI I think 21.1 firmware, I tried with the latest chirp-next, I tried with the latest egzumer.py plugin also the egzumer250000.py plugin, then I tried the compiled version of chirp on your github.  The cable was good.  I can get it to load and program egzumer .22 just fine but when I reset and reflash 21.1 I get that error in chirp no matter what.

mattfox27, 2024-04-11:
So do I just need to lower the baud in the python script?

kamilsss655, 2024-04-12:
No, it means it's a known bug and will be fixed in next releases.