kamilsss655/uv-k5-firmware-custom #179: [FEATURE] Improved Scanlist Management/Selection

VA7FGT, 2024-04-11:
I've been using this firmware for a few days and absolutely love it and the focus it puts on scanning and experimental features. 

While browsing the egzumer forks I came across [this firmware](https://github.com/AubsUK/uv-k5-firmware-custom) that integrates multiple scanlists and generally improved management/selection. 

It also has some options for power on scanning which is kind of cool! 

If possible would you consider merging @AubsUK s modifications into this one? With the fast scan and copy to VFO function this could be killer :)

Thanks!

AubsUK, 2024-04-11:
Appreciate you like the mods I've made @VA7FGT.  I'm slowly looking into utilising the mechanism the Spectrum Analyser uses into a Channel Sweep to speed up the scanning of memories, though I don't know if I'd do as good a job as this one (I haven't tried it yet, so no idea).  If they implement the changes, I might trash my repo in favour of it!

VA7FGT, 2024-04-11:
There's a ton of things that this firmware does really well for exploration compared to stock egzumer, give it a play and see how you like it (if you haven't already)! Worst case it might be a bit better suited for the scanning-first nature of your FW :P

kamilsss655, 2024-04-12:
Sounds like a great fork. I am not interested in porting it to this fw though.

closed 2024-04-12 by kamilsss655