kamilsss655/uv-k5-firmware-custom #189: [BUG] Deleted channels still appear in spectrum channel scan mode

darkhelmet46, 2024-06-22:
Description
When you delete a channel from memory using the radio keypad, and then enter Channel Spectrum mode, the radio still scans that frequency.  This also happens when removing a memory from a scan list.

**To Reproduce**
Steps to reproduce the behavior:
1. Program a frequency into a channel memory.
3. Add that memory to a scan list.
4. Enter spectrum channel scan mode.
5. Press 4 to switch to the scan list containing the channel memory.
6. Exit spectrum channel scan mode.
7. Delete the channel memory using the radio menus OR remove it from the scan list using the radio memories.
8. Enter spectrum channel scan mode.
9. Press 4 to switch to the scan list that contained the channel memory.
10. Note that the radio will still stop at that frequency and will display the frequency instead of the channel memory number, OR if you only removed it from the scan list it will still stop at that channel memory number.

Expected behavior
Expect is that spectrum channel scan mode will no longer try to stop on the frequency of the channel you deleted OR it will no longer try to stop on the channel memory number you removed from the scan list.

Details:
- Firmware version:  20.5

kamilsss655, 2024-06-22:
Unfortunately `v20.5` is not the latest, supported version.

Does this issue occur on `v21.4` ?

Does the issue go away after restarting the radio?

darkhelmet46, 2024-06-24:
Thanks for your reply!  Apologies, I was not aware there was a new
version.  i will update to the latest release, test, and let you know!
Thanks for getting back so quickly.

kamilsss655, 2024-06-27:
Closing this one as a stale issue. This can re-open once @darkhelmet46 reports back.

closed 2024-06-27 by kamilsss655

darkhelmet46, 2024-06-27:
Sorry for the late reply, I've had a busy week.  Also, sorry if I'm being
dumb but on github it shows 20.5 as the latest release... How do I get
21.4?
[image: image.png]