kamilsss655/uv-k5-firmware-custom #192: [FEATURE] Add optional documentation in Bahasa Indonesia

owl4ce, 2024-07-02:
**Is your feature request related to a problem? Please describe.**
Obviously, no. Especially, in ASEAN, probably many UV-K5 users might wants to read in their closely understandable language.

**Describe the solution you'd like**
I would like to suggest someone [who permitted] to add a formal documentation in Bahasa Indonesia. Thank you very much.

**Describe alternatives you've considered**
To **README.md**? or **wiki**?

**Additional context**
Here is for the sources.
1. https://github.com/owl4ce/uv-k5-id/releases/download/rr/Quansheng.UV-K5.Enhanced.Beginners.Guide.Bahasa.Indonesia.pdf
2. https://github.com/owl4ce/uv-k5-id

kamilsss655, 2024-07-02:
The documentation has English only language, and users can use translation service like Google Translate if needed.
I do not maintain multiple language versions as it would require additional work.

I am glad there is international audience for this firmware and appreciate your efforts with the translation though.

closed 2024-07-02 by kamilsss655

owl4ce, 2024-07-02:
Thanks for your great works! I mean, how about referencing to my repository, in wiki probably? I will synchronize the accurate translation with the latest wiki and my own firmware usability experience.