kamilsss655/uv-k5-firmware-custom #194: [FEATURE] send full payload of 30 characters in messages

sxxxse, 2024-07-25:
This is based on version 20.5    b655393ff7297575c354efafbb6816d817a8f625

i modified messenger.c and .h, to check if this could work, and it seems it works.

Also i applied the PMR lock patch, to only allow tx on PMR frequencies to that version 20.5.

[messenger30-v20.5.zip](https://github.com/user-attachments/files/16372524/messenger30-v20.5.zip)
^compiled firmware included, which seems to work on my uv-k5

sxxxse, 2024-07-26:
I noticed, in /ui/messenger.c the smallest font for rxMessage is moved 2 pixels to the right.
We could get additional space if we start at positon 0, and can make a whitespace after < or > or +.
Could then look like this. And i removed the Messenger title bar, to get additional space for 1 stored message.

[image]
[image]

[Archiv.zip](https://github.com/user-attachments/files/16391716/Archiv.zip)

kamilsss655, 2024-07-26:
Hey, looks cool!

Please mind that this is `Issues` section for this firmware. I'd suggest you create a custom fork where you can distribute the changes you have made.

Enjoy :)

closed 2024-07-26 by kamilsss655

sxxxse, 2024-07-27:
I thought, this may be an issue, because in the code the payload length is 30, so the last character was always unused,
which is inefficient, and maybe can cause a security risk in encryption.

I have had another idea, i want to let you know about encryption.

In manual, encryption key is described as 10 digits = 80 Bits, but in reality, its strength is not 80 Bits, but only 33.
(10^10 is ~about 2^33 ). We can get to about 61 Bits, by using 10 characters with T9 input from messenger for encryption key. log2( (26+26+10+5)^10).
It could work like this, to set encryption key in messenger:
Type message with 10 characters (or even more (30?)), which is stored in cMessage, but then not press M button to send, but another special button to store this enhanced encryption key, and then delete screen.

regards,
sxxxse