kamilsss655/uv-k5-firmware-custom #195: [BUG] DWR Keeps turning on automatically

WholesomeMeal, 2024-08-05:
Often after a scan, my radio will turn DWR on again. Usually have it set to DW. 

Firmware version: 20.5

UVK5 (8)

WholesomeMeal, 2024-08-06:
Unfortunately I don't have access to the early releases. I'll be happy to tell you once I have it.

WholesomeMeal, 2024-08-07:
Became a Patreon member to test this, yes it still happens. Sometimes when I turn my radio on, DWR is enabled. Despite setting it to DW.

Firmware: v21.4
UKV5(8)

WholesomeMeal, 2024-08-07:
If I scan through one of my lists, press exit (stop the scan), then turn it off - it will turn back on with **DWR enabled every second time.**

kamilsss655, 2024-08-07:
Ok, re-opening this one. :eyes:

kamilsss655, 2024-12-22:
Tested multiple times and could not reproduce with v.21.5. Closing this one. Can be re-opened if someone posts some more detailed steps / video of the issue.

closed 2024-12-22 by kamilsss655

WholesomeMeal, 2025-01-10:
Scan through saved memories, turn off then on. Repeat. 
When you turn the radio on again DWR will be enabled.

I have replicated this on 5 different radios.

v.21.7

WholesomeMeal, 2025-01-10:
I have a video of the issue but can't post here because the file size is too large.

kamilsss655, 2025-01-10:
It could be caused by some edge case related to the invalid EEPROM data on your device. That would explain why I could not reproduce it. Can you try factory reset and try to reproduce the issue?

WholesomeMeal, 2025-03-01:
Just tried the reset function (ALL) in the menu. Scanned through default memories. Issue is still there.

GlydetheSurfer, 2025-12-03:
having a bugfix locked behind a paywall is unfair to the people who dont have the money to support monthly or hate patreon with a burning passion (like me) because of the fact of paywalls. all the people with no money have to suffer with the bugs and inconveniences. it ISNT right.

WholesomeMeal, 2025-12-03:
@GlydetheSurfer The bug hasn't been fixed yet

GlydetheSurfer, 2025-12-04:
> > having a bugfix locked behind a paywall is unfair to the people who dont have the money to support monthly or hate patreon with a burning passion (like me) because of the fact of paywalls. all the people with no money have to suffer with the bugs and inconveniences. it ISNT right.
> 
> [@GlydetheSurfer](https://github.com/GlydetheSurfer) you’re welcome to try fixing it yourself, it doesn’t cost anything.

Its fine, I have my solution. >:)