kamilsss655/uv-k5-firmware-custom #202: No speaker (no audio at all) at Mesh Network firmware

bebudu, 2024-09-02:
**Describe the bug**
Speaker doesn't work (no beeps, no audio, nothing) with the Mesh Network v0.1, with Quansheng UV-K5(8)
**To Reproduce**
Steps to reproduce the behavior:
1. Flash Mesh Network v0.1 firmware to UV-K5(8)
2. Test audio

**Expected behavior**
No audio at all, no beeps nor voice.

kamilsss655, 2024-09-02:
What version of firmware are you using?
 I am not familiar with "Mesh Network v0.1 firmware".
Alternatively please provide the source link you got the firmware from.
Thanks.

bebudu, 2024-09-02:
The boot screen said "v 0.1" iirc, thus I mentioned it that way.
I am talking about the beta NUNU protocol firmware on your Patreon. https://www.patreon.com/posts/public-early-98088972

kamilsss655, 2024-09-02:
Ohh I see.

Yes, this works as intended as this is a demo version only, and not the latest currently supported one.

closed 2024-09-02 by kamilsss655