kamilsss655/uv-k5-firmware-custom #203: Piracy incident

randoassuser, 2024-09-05:
Hey nunu, Ive read about the piracy incident on your Patreon. I can understand your frustration but have you considered that individual is probably just frustrated with you closing the source of your project and even advertising it on an open source Platform? 

Also, whats the latest Patreon firmware? I am a bit confused with what the state of this project is and do you still maintain the open sourced repo? 

Best, Olli

randoassuser, 2024-09-05:
For context, I am not a Patreon member anymore and am considering whether to resubscribe or not as you have been focusing on the ESPR project which I was not too interested in

And I could imagine whoever pirated your software felt similar with the state of your firmware because they feel like you're wasting lots of potential by not publishing it. Publishing a premium distributed version is of course not the way to dispute this though lol, but just as a thought given to you, maybe communicate more clearly which features in the Documentation are premium features and where to get the version v21+ as that is not clear from this repo.

kamilsss655, 2024-09-05:
Thanks for your input.

closed 2024-09-05 by kamilsss655