kamilsss655/uv-k5-firmware-custom #207: [QUESTION] NOT BUG: Receiving messages sent from the HT via an SDR

AMDBartek, 2024-09-25:
Hello! First of all, I want to thank you for creating and maintaining this fork the firmware, I've been using it (with my own changes on top but nothing related to the messenger) and I like it a lot. Secondly, I'm sorry if this isn't the ideal place to contact you, but I've not found anywhere else (excuse me if it's painfully obvious). Anyway, with that out of the way, onto my question.

I have the radio and access the messenger to type a message and transmit it, and I am able to see and hear the transmission on my SDR (which is simply an RTL-SDR Blog v4). Then, I've remembered that `multimon-ng` has a AFSK1200 demodulator and tried it. which didn't work. I assume this is because the messages are packetized and encrypted before transmission, which is out of scope for `multimon-ng`.

So I decided to try and write my own script that uses my own radio framework (but compatible with open-source tools such as `rtl_fm`). It goes as follows:
```python
from scipy.signal import butter, lfilter
import numpy as np

# constants from the ESPRI code
AFSK_1200 = 1200
AFSK_2200 = 2200
SAMPLE_RATE = 25000
PAYLOAD_LENGTH = 30  # assuming the payload is 30 bytes
NONCE_LENGTH = 13  # nonce is 13 bytes, per espri code

# AGC parameters
DSP_AGC_MAX_VAL = 32767
DSP_AGC_MIN_VAL = -32767
DSP_AGC_TARGET_VALUE = DSP_AGC_MAX_VAL * 6 // 10
DSP_AGC_STEP = 10
DSP_AGC_SAMPLES_PER_STEP = 16000
DSP_AGC_MAX_GAIN = 255
DSP_AGC_MIN_GAIN = 1

# AGC class to handle gain control
class AGC:
    def __init__(self, initial_gain=1):
        self.current_gain = initial_gain
        self.samples_count = 0
        self.max_input_value = 0
        self.target_value = DSP_AGC_TARGET_VALUE
        self.samples_per_step = DSP_AGC_SAMPLES_PER_STEP
        self.gain_step = DSP_AGC_STEP

    def update(self, value):
        amplified_value = abs(value) * self.current_gain
        if amplified_value > self.max_input_value:
            self.max_input_value = amplified_value

        if self.samples_count < self.samples_per_step:
            self.samples_count += 1
        else:
            error = self.target_value - self.max_input_value
            if error > 0:
                self.current_gain = min(self.current_gain + self.gain_step, DSP_AGC_MAX_GAIN)
            else:
                self.current_gain = max(self.current_gain - self.gain_step, DSP_AGC_MIN_GAIN)

            self.max_input_value = 0
            self.samples_count = 0

        # prevent clipping
        if value > DSP_AGC_MAX_VAL / self.current_gain:
            return DSP_AGC_MAX_VAL
        elif value < DSP_AGC_MIN_VAL / self.current_gain:
            return DSP_AGC_MIN_VAL
        else:
            return value * self.current_gain

# AFSK demodulation class
class AFSKDemodulator:
    def __init__(self):
        self.samples_since_transition = 0
        self.samples_read = 0
        self.samples_window = SAMPLE_RATE // AFSK_1200  # AUDIO_AFSK_SAMPLESPERBIT equivalent
        self.last_sample = 0
        self.current_bit = 0
        self.transition_detected = False

    def demodulate(self, sample):
        self.samples_read += 1
        if not self.transition_detected:
            if sample == self.last_sample:
                self.samples_since_transition += 1
            else:
                self.transition_detected = True

        self.last_sample = sample

        if self.samples_read >= self.samples_window:
            if self.transition_detected:
                # Adjust sample window based on where the transition happened
                if self.samples_since_transition > (self.samples_window / 2):
                    self.samples_window = SAMPLE_RATE // AFSK_1200 + 5  # AUDIO_AFSK_SAMPLES_STEP equivalent
                else:
                    self.samples_window = SAMPLE_RATE // AFSK_1200 - 5
            else:
                self.samples_window = SAMPLE_RATE // AFSK_1200

            self.current_bit = 1 if sample > 0 else 0

            self.samples_read = 0
            self.samples_since_transition = 0
            self.transition_detected = False

        return self.current_bit

# butterworth filter
def butter_bandpass(lowcut, highcut, fs, order=5):
    nyquist = 0.5 * fs
    low = lowcut / nyquist
    high = highcut / nyquist
    b, a = butter(order, [low, high], btype='band')
    return b, a

def bandpass_filter(data, lowcut, highcut, fs, order=5):
    b, a = butter_bandpass(lowcut, highcut, fs, order=order)
    y = lfilter(b, a, data)
    return y

# apply AFSK demodulation on the provided raw audio data
def afsk_demodulate(audio_data):
    # Apply filters to isolate 1200Hz and 2200Hz signals
    filtered_1200 = bandpass_filter(audio_data, AFSK_1200 - 100, AFSK_1200 + 100, SAMPLE_RATE)
    filtered_2200 = bandpass_filter(audio_data, AFSK_2200 - 100, AFSK_2200 + 100, SAMPLE_RATE)

    agc = AGC()
    afsk = AFSKDemodulator()
    bitstream = []

    # demodulate audio data
    for sample_1200, sample_2200 in zip(filtered_1200, filtered_2200):
        sample = agc.update(sample_1200 - sample_2200)
        bit = afsk.demodulate(sample)
        bitstream.append(bit)

    return bitstream

# load the raw audio recorded with command rtl_fm -f <freq_in_mhz>M -s 25000 -l 150 > unencrypted-recording.bin
file_path = 'unencrypted-recording.bin'
with open(file_path, 'rb') as f:
    raw_audio_data = np.frombuffer(f.read(), dtype=np.int16)

# do demodulation
bitstream = afsk_demodulate(raw_audio_data)

# check the bitstream length and first few bits
print(len(bitstream), bitstream[:20])

# Decode the packet from the bitstream
def decode_packet(bitstream):
    """
    try extract the packet from the bitstream.
    the bitstream should contain header, payload, and nonce.
    """
    # assuming bitstream already in bytes
    byte_stream = []
    for i in range(0, len(bitstream), 8):  # convert bits to bytes
        byte = 0
        for j in range(8):
            if i + j < len(bitstream):
                byte |= (bitstream[i + j] << (7 - j))
        byte_stream.append(byte)

    # try extract header, payload, and nonce
    header = byte_stream[0]
    payload = byte_stream[1:1 + PAYLOAD_LENGTH]
    nonce = byte_stream[1 + PAYLOAD_LENGTH:1 + PAYLOAD_LENGTH + NONCE_LENGTH]

    return header, payload, nonce

# decode the packet from the bitstream
header, payload, nonce = decode_packet(bitstream)

# convert payload to string
message_body = bytes(payload).decode('utf-8', errors='ignore')
print(header, message_body)
```

But that only results in:
```
57344 [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1]
0 ?
```

Message Encryption was disabled in the menu. Anyway, any pointer in the right direction would be appreciated!

ifuchs, 2024-09-25:
@AMDBartek I would like to get messaging to work from a Mac using a serial connection to the UV-K5 using the Nunu messaging. The problem is that I connect the radio using an AIOC cable but I must not have the correct serial parms as I can't get the radio to respond to commands (e.g., get_info). Can you share how you connected your radio to your computer?

AMDBartek, 2024-09-25:
@ifuchs Yes, no problem. I am not sure how it'd be done on a Mac, but I assume it's similar on Linux. It worked for me out of the box using a Baofeng programming cable along with one I made by myself with a USB serial UART. The only thing I had to do was to set the baud rate to 38400 and everything went smoothly.

As for sending the messages, I am sending them using the keyboard on the UV-K5 itself and only receiving them with the RTL-SDR Blog V4 SDR (software defined radio) plugged into my computer. I am not sending the messages from my computer, only receiving them over the air.

ifuchs, 2024-09-25:
@AMDBartek Thanks. I saw that the SMS:text functionality will require a recompile with that flag set but before I did that I wanted to verify that I could in fact talk to the radio over the serial port. I just tried again with minicom and with the port set to 38400 8N1 but it does not show as being online. The USBmodem shows up in /dev/tty. I am missing something...

ifuchs, 2024-09-25:
@AMDBartek does the UV-K5 have to be in programming mode to have it communicate in this way? I tried and some gibberish comes out but I still cannot talk to it.

AMDBartek, 2024-09-25:
@ifuchs No prob. I just tried the SMS over serial and it is a bit unreliable, but always seems to work the first time. The radio has to be powered on (not in programming mode) and then you connect. For `minicom`, you can run `minicom -s` to configure it. Make sure the Bps/Par/Bits is set to `38400 8N1` and that has the correct settings.

Alternatively, you could try `echo`ing the command directly into the radio like so via the shell:
```sh
# Set up serial port
stty --file=/dev/ttyUSB0 38400 -parenb cs8 -cstopb
# Send the SMS command to the radio
echo -n "SMS:hello" > /dev/ttyUSB0
```

Obviously on macOS, the device names (and potentially command formats) may be different but from what I know, it should work the same way if you run `zsh` or `bash`. The TX/RX LED on the radio will turn red if it is sending the message and you'll see it pop up in the messenger on the screen if you have it open.

This isn't what I'm trying to achieve personally, but I'm happy to help out as best as I can.

ifuchs, 2024-09-25:
@AMDBartek I have not yet recompiled the fw to add the SMS functionality as I wanted to confirm that serial comm worked before I make matters more complicated. I have connected the radio with the AIOC cable which creates the USB modem (this works fine for programming/upgrades). I have tried with minicom and with screen command set to 38400 8N1 but so far I can't get the radio to respond to any commands (e.g., GET_INFO).

AMDBartek, 2024-09-25:
@ifuchs Honestly, I'm not entirely sure, sorry. I personally never used the other serial commands so I'm not sure. But if you want to check if serial communications work before deciding to recompile the firmware, you could try CHIRP.

You would probably need this driver: https://github.com/kamilsss655/uvk5-chirp-driver

You could also recompile the firmware and see, it only takes less than 5 seconds on my PC so you don't have to wait a long time or anything like that.

ifuchs, 2024-09-25:
@AMDBartek I just tried CHIRP and it works fine. Just can't get it to work from command line :-(

AMDBartek, 2024-09-25:
@ifuchs Not sure then, perhaps you'd have to wait until the owner of the repository replies. Or I guess you could just try rebuilding with the flag set just to see, maybe it does work. In my test, `echo`ing it directly into the serial interface file (`/dev/ttyUSB0` or macOS equivalent) worked the best, I didn't have luck with `minicom` either for some reason, just showed as offline.

ifuchs, 2024-09-25:
@AMDBartek If you echo with "GET_INFO" do you get a response? I tried with stty and echo and did not get anything back.

AMDBartek, 2024-09-25:
@ifuchs Nope, because that's not how it works. What you'd need to do is open a second terminal and run `cat /dev/ttyUSB0` and in your first terminal, run `echo -n "GET_INFO" > /dev/ttyUSB0`. Of course, replace `/dev/ttyUSB0` with your respective device.

Try that and see if you get anything back from the radio, you gotta be listening for the message to get one, let me know if it works, I'm curious too!

ifuchs, 2024-09-25:
@AMDBartek Nothing came back on the other terminal session using cat.

AMDBartek, 2024-09-25:
@ifuchs Oops, forgot to mention, you'd probably have to run `stty --file=/dev/ttyUSB0 38400 -parenb cs8 -cstopb` first to ensure that the correct baudrate is set. The early firmware from Patreon apparently has a different baudrate, so keep that in mind if you're using it, if not it'll be 38400.

ifuchs, 2024-09-25:
I did that in the terminal that sent the echo. No dice. Minicom also shows Offline for me as well. So I have struck out with Screen and Minicom commands as well as stty, echo. But Chirp works fine so the USB modem must be working. Could it be that the speed is not 38400? I tried with screen and mincom at all of the avaialble speeds but something else must be set wrong.

AMDBartek, 2024-09-25:
@ifuchs Potentially yes actually, try a baudrate of 115200 and see how that goes.

ifuchs, 2024-09-25:
@AMDBartek I just tried 115200 with minicom and stty/echo with no response.

AMDBartek, 2024-09-25:
@ifuchs Ah, that's unfortunate. I'm out of ideas personally, sorry about that. I guess only thing that I can think of is compiling the firmware with messenger UART enabled and giving it a shot anyway, worst comes to worst, it just won't work, I guess.

ifuchs, 2024-09-25:
@AMDBartek From what the docs say, I thought that was just for the SMS functionality as opposed to all serial comm. I may try with another totally unmodified UV-K5 just to see if serial works there (or re-flash this one with another fw).
Thanks for your help.

kamilsss655, 2024-09-26:
@AMDBartek that's an ambitious attempt, and while I am do not have specific need for this I might suggest adjusting the window side:
```
self.samples_window = SAMPLE_RATE // AFSK_1200 + 5
```
either by adjusting the `SAMPLE_RATE` or the `+ 5` by a few %. It might be the case that the timing clocks of the devices are not exactly matched and/or what you think they are. If that doesn't help then you'd have to do more explorative work on your own. Good luck.

PS: @AMDBartek, @ifuchs I am closing this issue, however you may continue your conversation here.

closed 2024-09-26 by kamilsss655