kamilsss655/uv-k5-firmware-custom #211: [BUG] Please use a title that makes sense

dfx12, 2024-10-28:
**Describe the bug**
With CTCSS Tone set,  no audio  on Receive for Tones 94.7-103.5 - Green light flashes indicating receive but no audio through the speaker

Tested on two indentical UV K5(8) both running v20.5

Appears only to be an RX issue - testing to a 3rd Radio (Baofeng uv 17) TX tone appears to be sent correctly as this Radio with 
the same Tone set can receive without issue

Tones above and below this range work

May be related to :

"ENABLE_MESSENGER_FSK_MUTE          := 1       mutes speaker once it detects fsk sync word (might cause unintentional mutes during ctcss rx)"

but I am unable to compile firmware myself with the flag set to 0 to test

**To Reproduce**
Steps to reproduce the behavior:
1.  Set CTCSS Tone between 94.7-103.5
2. transmit on another device with  tone set to the tone above
3.  Green light on receiving unit with flash but no audio will come through 
4. See error

**Expected behavior*
Audio to be played through speaker

**Pictures/video**
If applicable, add pictures or video to help explain your problem.
To insert a picture (.jpg), you can simply drag it to this window.

**Details:**
- Firmware version: 20.5

kamilsss655, 2024-12-22:
Does the issue exist on v.21.5 ?

kamilsss655, 2024-12-23:
Cannot replicate on v.21.5 with `97.4Hz` set for rx and tx ctcss.

closed 2024-12-23 by kamilsss655