kamilsss655/uv-k5-firmware-custom #214: [FEATURE] Receiver only mode

kamilsss655, 2024-12-16:
**Is your feature request related to a problem? Please describe.**
For some audience it could be useful to have this device operate in `RX` only mode. To alleviate concerns of the accidental press of the `PTT` button there should be a way of turning the `TX` off.

**Describe the solution you'd like**
Either some sort of menu option like `TxMode` that can be disabled or enabled or have the `PTT` button function be programable just like with other buttons, so user can still use this button for other functionalities based on their preference.

kamilsss655, 2024-12-22:
Oh this can already by done by using hidden menu entry `F Lock` and setting it to `DISABLE ALL`

closed 2024-12-22 by kamilsss655

ErikS-web, 2024-12-23:
See [Wiki menu-item 54](https://github.com/kamilsss655/uv-k5-firmware-custom/wiki/90-%E2%80%90-20-%E2%80%90-Menu#hidden-menu)