kamilsss655/uv-k5-firmware-custom #216: [FEATURE] Improve lock function

kamilsss655, 2024-12-22:
**Is your feature request related to a problem? Please describe.**
Based on some of the user reports currently when device is locked user can still use `FN1` and `FN2` buttons (along with `PTT`).

These function buttons can be assigned different functions, including changes to channel width etc.

This prevents the usage in the scenario when 1 preconfigured radio is handed of to an unexperienced user or a child when basic communication is needed. In such scenario one would only desire the other  operator to have access to `PTT` without risking unintentional reconfiguration of the communication parameters.

**Describe the solution you'd like**
Locked state should only allow `PTT` button to be used.

kamilsss655, 2024-12-22:
Locked state already persists through reboots, which is a good thing in such scenario and requires no changes.

wd4sgu, 2024-12-22:
This looks like really nice work. Is there anywhere that I can download a compiled .bin file? I do not have the tools to make it myself. Thanks.

kamilsss655, 2024-12-23:
Thank you. Most recent releases are available at [patreon.com/kamilsss655](https://patreon.com/kamilsss655)

kamilsss655, 2024-12-23:
fixed in 21.6.
also fixed auto lock feature while at it.

closed 2024-12-23 by kamilsss655